genodelabs/genode #4152: vbox5: numlock ineffective in Linux guest with X server

jschlatow, 2021-05-14:
I just noticed that my Linux guest does not fully adapt to numlock state changes. It appears to be an issue of the guest system though. Nonetheless, I consider it worth mentioning here for documentation. Numlock is implemented in Sculpt by dynamic remapping of the keypad keys, hence there is no need to propagate the numlock state to the VM...in principle. In my Linux VM, however, the X server seems to remap the keypad keys itself so that numlock changes do not have any practical effect. If I have a look at the input device with evtest, the reported keycodes perfectly change depending on the global numlock state. xev, however, still reports KEY_LEFT instead of KEY_KP4 for instance, irrespective of whether numlock is disabled or enabled.

nfeske, 2021-05-17:
It seems that the X server is more clever than we want it to be. Do you have an idea how to go about it?

jschlatow, 2021-05-17:
The following `/config/numlock_remap` seems to solve it:

```
<config>
        <input name="numlock_enabled" rom="numlock" node="numlock">
                <attribute name="enabled" /> </input>

        <output node="remap">
                <if>
                        <has_value input="numlock_enabled" value="no"/>
                        <then>
                                <inline>
                                        <key name="KEY_KP0"   to="KEY_INSERT"/>
                                        <key name="KEY_KP1"   to="KEY_END"/>
                                        <key name="KEY_KP2"   to="KEY_DOWN"/>
                                        <key name="KEY_KP3"   to="KEY_PAGEDOWN"/>
                                        <key name="KEY_KP4"   to="KEY_LEFT"/>
                                        <key name="KEY_KP5"   to="KEY_RESERVED"/>
                                        <key name="KEY_KP6"   to="KEY_RIGHT"/>
                                        <key name="KEY_KP7"   to="KEY_HOME"/>
                                        <key name="KEY_KP8"   to="KEY_UP"/>
                                        <key name="KEY_KP9"   to="KEY_PAGEUP"/>
                                        <key name="KEY_KPDOT" to="KEY_DELETE"/>
                                </inline>
                        </then>
                        <else>
                                <inline>
                                        <key name="KEY_KP0"   to="KEY_0"/>
                                        <key name="KEY_KP1"   to="KEY_1"/>
                                        <key name="KEY_KP2"   to="KEY_2"/>
                                        <key name="KEY_KP3"   to="KEY_3"/>
                                        <key name="KEY_KP4"   to="KEY_4"/>
                                        <key name="KEY_KP5"   to="KEY_5"/>
                                        <key name="KEY_KP6"   to="KEY_6"/>
                                        <key name="KEY_KP7"   to="KEY_7"/>
                                        <key name="KEY_KP8"   to="KEY_8"/>
                                        <key name="KEY_KP9"   to="KEY_9"/>
                                </inline>
                        </else>
                </if>
        </output>
</config>
```

chelmuth, 2021-05-17:
This is the solution I hoped for as it extends our approach to *hide* the numpad from the VMs to the max.

chelmuth, 2021-05-17:
On the other, the approach also rules out all applications that make use of the numpad for special-function mapping (Blender?). So, it may be better to fix this in the VM by enforcing NumLock=on in some way (e.g., `setxkbmap -option numpad:mac`). If I'm not mistaken, this is the way Windows works.

jschlatow, 2021-05-17:
@chelmuth  `numpad:mac` also does the trick.

closed 2021-06-02 by jschlatow