genodelabs/genode #4157: foc/imx6: usb_hid_raw gets stuck with bufferable DMA memory

chelmuth, 2021-05-18:
While fixing an unaligned read in #4094, I discovered the usb_hid_raw test fails on foc/imx6q_sabrelite since months. This stems from our approach to map normal memory used for DMA buffers *bufferable*, which is required by ported linux drivers that expect support for unaligned read operations on buffers. In case of usb_hid_raw, it is USB descriptor parsing in the host driver. Unfortunately, I had not enough time to compare the resulting memory attributes of hw and foc as the latter uses TEX remapping on ARMv6+, which introduces a level of indirection I'm unexperienced with.

chelmuth, 2021-05-18:
Commit f24d9edc2566983f470265e25f7e40270ee29c98 does a rude quirk to make the test work again and showcases that some uncached normal memory mappings require different memory attributes than applied by foc with `L4_FPAGE_BUFFERABLE`.

skalk, 2021-06-04:
@chelmuth: I can confirm that the additional `L4_FPAGE_BUFFERABLE` that you've enabled to map uncached memory under Fiasco.OC leads to the right page-table settings. I have checked the state before and after your commit using a JTAG debugger connected to the i.MX 6Q Sabrelite. Without the `L4_FPAGE_BUFFERABLE` flag, the memory gets described as "strongly ordered", which should not be used for normal, uncached RAM. With the addition of the flag, it is now set as uncached, normal memory.

chelmuth, 2021-06-04:
Thanks for recording your findings here. Regarding my tests in f24d9edc2566983f470265e25f7e40270ee29c98, we came to the conclusion that the remaining issue with DMA buffers were *bufferable* does not work stems from inefficient barrier implementations in dde_linux.

chelmuth, 2023-06-06:
Close as fixed.

closed 2023-06-06 by chelmuth