genodelabs/genode #4158: USB: add webcam support

ssumpf, 2021-05-18:


ssumpf, 2021-05-18:
The commits above add USB webcam support to Genode. 
3562c7b is a test that uses a dynamic `init` to spawn and kill the webcam app and is not necessary for staging.

ssumpf, 2021-05-18:
@chelmuth: 784dd3c your suggestions. As said 3562c7b remains strictly for testing  until I find out how to connect to `rom_reporter` correctly in order to get rid of most of the run scripts components.

chelmuth, 2021-05-18:
Merged all commits ex. run script to staging. In my opinion, the use of *drivers_managed* is not a must-have. I'd rather love to get a depot package for easy deployment in Sculpt.

chelmuth, 2021-05-19:
The changes to libyuv break VirtualBox depot archives vbox5-nova, vbox5, and vbox6.

```
make[5]: *** No rule to make target 'jpeg.lib', needed by 'libyuv.lib'.
make[5]: Target 'all' not remade because of errors.
make[4]: *** [Makefile:302: gen_deps_and_build_targets] Error 2
make[4]: Failed to remake makefile 'Makefile'.
make[4]: Target '_all' not remade because of errors.
make[3]: *** [/data/genode/tool/depot/mk/build_bin_archive:201: /data/depot/genodelabs/bin/x86_32/vbox5-nova/2021-05-18.build/bin] Error 1
make[3]: Target 'vbox5-nova/2021-05-18' not remade because of errors.
make[2]: *** [/data/depot/var/build.mk:12: genodelabs/bin/x86_32/vbox5-nova/2021

ssumpf, 2021-05-19:
@chelmuth: Ok, will have a look.

ssumpf, 2021-05-19:
@chelmuth: cfd0b74 fixes the VirtualBox packages. I really missed this one.

chelmuth, 2021-05-19:
Merged as 3536576e40b2870e5e7010fb1dc78254bd28cc0d. It required digging to understand why jpeg is necessary in *used_apis* but not libyuv. libyuv (as well as qemu-usb) is packaged with vbox5/6.

ssumpf, 2021-05-19:
@chelmuth: Because `libyuv` should have its own api package.

chelmuth, 2021-05-19:
Autopilot reveals that vbox scenarios now fail with the following.

```
[init -> vbox1] Error: Could not open ROM session for "jpeg.lib.so"
```

chelmuth, 2021-05-19:
`make -C build/x86_64 app/usb_webcam` fails with the following.

```
  Program app/usb_webcam/usb_webcam
    COMPILE  main.o
/.../genode/repos/libports/src/app/usb_webcam/main.cc:124:7: error: ‘class Webcam’ has pointer data members [-Werror=effc++]
  124 | class Webcam
      |       ^~~~~~
/.../genode/repos/libports/src/app/usb_webcam/main.cc:124:7: error:   but does not declare ‘Webcam(const Webcam&)’ [-Werror=effc++]
/.../genode/repos/libports/src/app/usb_webcam/main.cc:124:7: error:   or ‘operator=(const Webcam&)’ [-Werror=effc++]
/.../genode/repos/libports/src/app/usb_webcam/main.cc:131:24: note: pointer member ‘Webcam::_handle’ declared here
  131 |   uvc_device_handle_t *_handle  { nullptr };
      |                        ^~~~~~~
cc1plus: all warnings being treated as errors
```

ssumpf, 2021-05-19:
@chelmuth: b916321 fixes the compiler error. The USB session problem takes longer than expected tough.

alex-ab, 2021-05-20:
> Autopilot reveals that vbox scenarios now fail with the following.
> 
> ```
> [init -> vbox1] Error: Could not open ROM session for "jpeg.lib.so"
> ```

@chelmuth: please add 76d80097273bd5b0a1e305030575c299de2c717b to staging

nfeske, 2021-05-20:
Thank you. Merged to staging.

ssumpf, 2021-05-20:
9bc0f87 handles the case where frames are incomplete for YUV to ARGB conversions. For MJPEG to ARGB this it not a problem, since the conversion function takes the size of the frame.

nfeske, 2021-10-14:
Merged to master and featured by Sculpt 21.10