genodelabs/genode #4160: rump: compiler warnings

m-stein, 2021-05-18:
When I create the bin/x86_64/rump archive on staging, I observe the following compiler warnings:
```
  Library vfs_rump
    COMPILE  random.o
    COMPILE  vfs_rump.o
/home/lypo/genodelabs/genode/depot/mstein/src/rump/2021-05-18/src/lib/vfs/rump/vfs_rump.cc: In member function ‘Vfs::File_io_service::Read_result Vfs::Rump_file_system::Rump_vfs_dir_handle::_finish_read(const char*, dirent*, Vfs::Directory_service::Dirent&)’:
/home/lypo/genodelabs/genode/depot/mstein/src/rump/2021-05-18/src/lib/vfs/rump/vfs_rump.cc:214:52: warning: narrowing conversion of ‘(s.stat::st_mode & 64)’ from ‘__mode_t’ {aka ‘unsigned int’} to ‘bool’ [-Wnarrowing]
  214 |                           .executable = (s.st_mode & S_IXUSR) };
      |                                         ~~~~~~~~~~~^~~~~~~~~~
/home/lypo/genodelabs/genode/depot/mstein/src/rump/2021-05-18/src/lib/vfs/rump/vfs_rump.cc: In constructor ‘Vfs::Rump_file_system::Rump_file_system(Vfs::Env&, const Genode::Xml_node&)’:
/home/lypo/genodelabs/genode/depot/mstein/src/rump/2021-05-18/src/lib/vfs/rump/vfs_rump.cc:448:12: warning: unused variable ‘available’ [-Wunused-variable]
  448 |     double available = factor * stats.f_bsize * stats.f_bavail;
      |            ^~~~~~~~~
/home/lypo/genodelabs/genode/depot/mstein/src/rump/2021-05-18/src/lib/vfs/rump/vfs_rump.cc:449:12: warning: unused variable ‘total’ [-Wunused-variable]
  449 |     double total = factor * stats.f_bsize * stats.f_blocks;
      |            ^~~~~
/home/lypo/genodelabs/genode/depot/mstein/src/rump/2021-05-18/src/lib/vfs/rump/vfs_rump.cc: In member function ‘virtual Vfs::Directory_service::Stat_result Vfs::Rump_file_system::stat(const char*, Vfs::Directory_service::Stat&)’:
/home/lypo/genodelabs/genode/depot/mstein/src/rump/2021-05-18/src/lib/vfs/rump/vfs_rump.cc:712:43: warning: narrowing conversion of ‘(sb.stat::st_mode & 64)’ from ‘__mode_t’ {aka ‘unsigned int’} to ‘bool’ [-Wnarrowing]
  712 |                 .executable = (sb.st_mode & S_IXUSR) },
      |                               ~~~~~~~~~~~~^~~~~~~~~~
    MERGE    vfs_rump.lib.so
```

nfeske, 2021-05-18:
I remember having see similar warnings and think the proper fix is replacing `(sb.st_mode & S_IXUSR)` by `(sb.st_mode & S_IXUSR) != 0`.

m-stein, 2021-05-19:
Fixed by 7fd308acec. And produced also a fixup fbdd9f3c1b for #4032.

commit d5c9f435 (2021-05-19): rump: fix compiler warning

Fixes two warnings of this type:

! src/lib/vfs/rump/vfs_rump.cc:214:52: warning: narrowing conversion of
! ‘(s.stat::st_mode & 64)’ from ‘__mode_t’ {aka ‘unsigned int’} to ‘bool’
! [-Wnarrowing]
!
!  214 |                           .executable = (s.st_mode & S_IXUSR) };
!      |                                         ~~~~~~~~~~~^~~~~~~~~~

Fixes #4160
closed 2021-05-28 by chelmuth