genodelabs/genode #4163: sculpt: add support for global bashrc

jschlatow, 2021-05-19:
There are already multiple applications running bash  (inspect view, system shell). Similar to the global vimrc, I suggest adding a global bashrc for a more consistent look and feel. The default bashrc may contain aliases such as `cp='cp -i'` to prevent accidental overwriting of existing files.

jschlatow, 2021-05-19:
Commit 46e4d33 adds the default bashrc to the sculpt scenario. Packages (e.g. system_shell) still require adaptation to make use of it.

nfeske, 2021-05-19:
Even though I see the practical value of your suggestion for the current version of Sculpt, this is not the direction I anticipate in the longer term. In fact, I regard the unixy parts of the Sculpt base image as an interim solution. The inspect shell will eventually go away once the file browser has left its infancy. The system shell (as a component) will prevail of course. The following two links shed more light on my reasoning.

https://lists.genode.org/pipermail/users/2021-January/007475.html

https://genode.org/documentation/release-notes/20.02#Redesign_of_the_administrative_user_interface_of_Sculpt_OS

Hence, when speaking of system-wide consistency, I'd like move the focus away from Sculpt's leitzentrale but towards components deployed in the runtime. E.g., similar to how the fonts fs ties together GUI components with respect to font rendering, there could be a central component that hands out dot files to other components. So when making tweaks there, all users will adapt.

jschlatow, 2021-05-19:
Understood. Following the scheme of the fonts fs would be more elegant.

jschlatow, 2021-05-21:
@nfeske, I force pushed a new commit that only contains the patch of bash to change the default bashrc path.

The bashrc and vimrc can be easily pulled from a filesystem session. I created a `dotfiles_fs` pkg with Goa that populates an arbitrary file system with a default bashrc and the global vimrc (see [goa-pkgs](https://github.com/jschlatow/goa-pkgs)). Works like a charm.

nfeske, 2021-06-04:
I have merged your commit to staging now.

nfeske, 2021-08-31:
Fixed in master.

closed 2021-08-31 by nfeske