genodelabs/genode #4164: unable to cycle through windows if no window is currently focused

jschlatow, 2021-05-19:
I've become a bit annoyed by the fact that I need to hover into the first window I open using the mouse rather than tabbing into it with the keyboard. Currently, if no window is focused the focus history next() method will always return an invalid window id even if the history is not empty.

jschlatow, 2021-05-21:
I'm afraid this issue might need a more sophisticated fix. I just observed that menu_view windows (e.g. file_vault, top_view) behave differently. I haven't dug into it yet, though it appears to me that elements such as buttons unfocus the window and reset the window layouter to a state in which the focused window id cannot be found in the history. In consequence, with my previously proposed change, the first window is focused and raised (which is not necessarily the one I'm currently working in). I'll have a close look next week.

jschlatow, 2021-05-26:
As it turned out, commit 47f996a does not cause the observed misbehaviour of menu_view windows. I created a separate issue for this: #4176.

nfeske, 2021-05-26:
Do you suggest to merge your fix? I have no reservation.

jschlatow, 2021-05-26:
Yes, the fix can be merged.

nfeske, 2021-05-26:
It's my pleasure.

closed 2021-05-28 by chelmuth