genodelabs/genode #4174: falkon: second window when dragging text

m-stein, 2021-05-26:
When I select a text on a webpage in Falkon (using themed_wm) and then drag it with the mouse, a second window with a black background pops up behind my Falkon. This new windlow is quite big and contains the selected text stretched to the window size and in strange bright colors. As soon as I release the mouse button, the additional window disappears. It seems, that it doesn't matter which text I select or how much of it.

nfeske, 2022-04-13:
BTW, this also happens when dragging an image.

I think Qt implements the drag operation by showing the dragged item as a special top-level window that is ought to follow the mouse position. However, using the window manager, the position of top-level windows cannot be controlled by applications.

To address this issue in way natural to Genode, we'd need to change the Qt implementation by propagating the image of the dragged item as part of the mouse shape. The latter is actually controlled by Qt already. But I'm afraid that this change may still be quite invasive in Qt. Maybe it's best to simply disable the creation of a window for the dragged item? Sooner or later, we may consider supporting drag and drop in the nitpicker GUI server. But that's not planned for now.

m-stein, 2022-05-31:
It seems that there is no plan or pressing need to solve this issue. I'd therefore close it. Any objections?

m-stein, 2022-05-31:
If anyone would like to take over in this matter, I'd suggest he/she re-opens the issue.

closed 2022-05-31 by m-stein