genodelabs/genode #4176: themed_wm: window focus not maintained in menu_view dialogs

jschlatow, 2021-05-26:
When testing the file vault, I noticed an issue with the window focus. In order to reproduce this, I launch file vault and falkon browser and connect both to my themed_wm. When I drag the windows such that falkon is stacked below the file vault, it sometimes happens that the focus of the file vault window is not maintened when I move the cursor within the window boundaries. This can be seen by looking at the title bar that changes into a darker shade even though the cursor is still in the file vault window. When I now click a menu item, the falkon browser is raised.

It seems that it may require a certain sequence of interactions to get into this state. It does not occur every time but I'm able to reproduce this by hovering the cursor back and forth between the title bars of both windows.

nfeske, 2021-05-26:
Today I have seen this happen, too. But I have not yet found a particular interaction pattern that reproduces it.

jschlatow, 2021-06-17:
I think I found a pattern that reproduces this. The starting point is to position the file_vault and falkon windows partially overlapping as shown in the screen shot:
[image]

Then I perfrom the following sequence:

1. With the file_vault in foreground, move the pointer into the file_vault window.
2. Press and hold LEFTMETA while moving the pointer out of the file_vault boundaries into the falkon window.
3. Press TAB and release both, TAB and LEFTMETA.
4. falkon should now be raised to top:

[image]

5. Now, hit and release LEFTMETA+TAB again to raise file_vault again.

[image]

You'll now have the situation that even if the pointer is within the file_vault window, the falkon window remains focused.

Interestingly, I can reverse this by starting with the pointer in falkon, pressing and holding LEFTMETA, moving the pointer into file_vault, hitting TAB and releasing all keys again.

jschlatow, 2021-06-17:
I just noticed that I can also reproduce this without the TAB key, i.e. only by pressing and holding LEFTMETA and releasing when the pointer was moved to another window.

@nfeske, do you have an idea what might cause this behaviour?

chelmuth, 2021-06-18:
I think I could reproduce this behavior with two terminals (system shell instances in my case).

First,  I followed @jschlatow's instruction using **LEFTMETA** without **TAB**. If I afterwards hover the title bar of the upper window key events reach this window while the events are routed to the lower window if I hover the window contents.

chelmuth, 2021-06-18:
As always it's also possible to force input-event routing to the hovered window by pressing and holding the mouse button while typing.

Note, I'm using sculpt-21.03b currently.

nfeske, 2021-10-14:
Since the commits above fix the problem described by @jschlatow, I'm closing this issue. The discrepancy between nitpicker's per-pixel input mask and the window manager's notion of rectangular windows is a separate topic that we may address later.