genodelabs/genode #4177: dde_bsd: download URL for OpenBSD is 404 (can't build audio_drv)

ttcoder, 2021-05-26:
Apologies if this was a known issue at the time that got fixed in the meantime ; though from what I remember of reading the release notes back then, and in the intervening time, Genode's dde_bsd is still (?) at OpenBSD 6.6 ; so it might be that the problem is real and I'm just the first one to notice it.

Anyway -- I'm currently trying to build Genode from source (20.2 as of today, I was at 18.11 a few days ago, catching up fast ^^), and invoking "prepare_port dde_bsd" fails immediately with a 404-style error, on the ftp-aachen server. Is that reproducible/confirmed at your end ?

I had the same problem in 19.x in the past few days, but those older revs use an older file/URL (OpenBSD 5.9) so I assumed it was just gone for good and the problem would be solved with the updated driver in Genode 20.02... Dang.

tomga, 2021-05-26:
I'm curious why you're trying so old version. Actually I initially assumed last stable version 21.02 and was going to write that it works for me :-)

Anyway it was fixed in 6ea628195fc750af37df48bb81cdfcb521ef85c7.

ttcoder, 2021-05-26:
Thanks much tomga! I searched the ticket DB for "bsd" and "audio_drv" before filing this but didn't think of checking the commit list. Will close this tomorrow I think.

For 3 long (very, very long) years I was stuck at 18.11 for various reasons, now updating my project step by step to account for changes that intervene at each quarterly release. Will post to my micro "blog" and/or the ML when I'm done, though it will make for kind of boring reading <g>

ttcoder, 2021-05-27:
It downloads and builds indeed, closing.

(FWIW, latest as of this morning is that I'm stopped from upgrading to 20.05 by a change in libc in 20.02 but it won't stop me long I think, gotta soldier on until I reach 21.02)

closed 2021-05-27 by ttcoder