genodelabs/genode #4178: fec nic driver: compiler warnings

m-stein, 2021-05-27:
Compiling the FEC NIC driver on staging gives me the following warnings: 
```
lx_emul.cc:711:13: warning: taking address of packed member of ‘page’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  711 |  atomic_inc(&page->_count);
      |             ^~~~~~~~~~~~~
lx_emul.cc:730:13: warning: taking address of packed member of ‘page’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  730 |  atomic_set(&page->_count, 1);
      |             ^~~~~~~~~~~~~
lx_emul.cc:755:27: warning: taking address of packed member of ‘page’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  755 |  if (!atomic_dec_and_test(&page->_count))
      |                           ^~~~~~~~~~~~~
```

m-stein, 2021-06-02:
I don't have a good idea how to fix this, but at the same time, I think, that we shouldn't accept compiler warnings in non-contrib code.

m-stein, 2021-06-28:
I have created two approaches to this issue. We should merge only one of them:
3e4234bfbd Approach #2 "fec nic driver: fix compiler warnings"
0cc2e454f1 Approach #1 "fec nic driver: fix compiler warnings"

chelmuth, 2021-06-28:
I vote for Approach 2 as this data is never accessed by the device driven and, thus, removing `packed` should be ok.

m-stein, 2021-06-28:
Fine with me :)

commit 677eaffb (2021-06-29): fec nic driver: fix compiler warnings

Fixes #4178
commit 388e2a0e (2021-07-28): fec nic driver: fix compiler warnings

Fixes #4178
closed 2021-07-28 by chelmuth