genodelabs/genode #4188: tool

nfeske, 2021-06-07:


nfeske, 2021-06-07:
@skalk the commits linked above contain my suggestions.

skalk, 2021-06-15:
@nfeske: Thank you very much for your careful improvements. Except for commit "show objs for variables", see my comment inline, I'm perfectly fine with the changes.

skalk, 2021-06-15:
Unfortunately, commit "out-of-tree build dir" breaks the tool's functionality for me. None of the function definitions can be generated anymore with a non-out-of-tree built Linux kernel.

nfeske, 2021-06-15:
I'm sorry for this regression and will look into it.

skalk, 2021-06-15:
No matter. I've created a fixup commit feb53a133ed1771562c8d0559433882e73e8b2b8 on my merge to staging branch, and re-applied the following commits of yours too.

nfeske, 2021-06-17:
Thank you @skalk for the careful review and the curation of the commit series. I merged the commits of your merge_to_staging branch to staging now.

chelmuth, 2021-07-28:
Merged to master.

closed 2021-07-28 by chelmuth