genodelabs/genode #4191: Remove sandbox and vfs symbols from os api

jschlatow, 2021-06-10:
With commit 89f813f1133aa1473ab796cbbaf28093d294a0b3, the sandbox and vfs symbols were added to the os api. This has the side effect that components developed in Goa will be linked against both libraries and thus require sandbox.lib.so and vfs.lib.so at runtime.

In an offline discussion with @nfeske, we decided it would be cleaner to introduce a distinct sandbox api. A vfs api containing the corresponding symbols is already present.

nfeske, 2021-06-17:
Thank you for the commit, which I merged to staging now.

It took me a while to make up my mind regarding the move of the header from os/sandbox.h to sandbox/sandbox.h but I think to probably more intuitive that way.

Just for the record, the VFS front-end API resides in include/os/vfs.h, deliberately not in include/vfs/. The latter location is only meant for the use by VFS plugins, not by users of the VFS. To attain consistency between the sandbox.h and vfs.h, we may consider renaming incude/os/vfs/ to something else (I have not good idea right now) and rename os/vfs.h to vfs/vfs.h. But there is no need to hurry.

jschlatow, 2021-06-17:
Thanks for the clarification about the VFS. What puzzles be then though, is that the current VFS api archive does not contain include/os/vfs.h. It is included in the os api though, which is probably why this has not popped up as an error yet.

Nonetheless, I like your suggestion of moving the front-end API to include/vfs for the sake of consistency. I'll create a separate issue for this. Do you also think that we may consider separating the front-end API and plugin API into two api archives?

nfeske, 2021-06-22:
> What puzzles be then though, is that the current VFS api archive does not contain include/os/vfs.h. It is included in the os api though, which is probably why this has not popped up as an error yet.

Since any target that uses the VFS API has to depend on the os API anyway, the os/vfs.h is always present. It's not accidental. I rather regarded the os/vfs.h frontend to be part of the os API. But since you identified the need to remove the symbol files from the os API archive (for the use of the os API by goa apps) and os/vfs.h is closely related to lib/symbols/vfs, the need for a more clear-cut split has became much more clear now.

> Do you also think that we may consider separating the front-end API and plugin API into two api archives?

This would be the cleanest solution.

nfeske, 2021-08-31:
Fixed in master.

closed 2021-08-31 by nfeske