genodelabs/genode #4193: Inconsistencies in build system re. lib/import/import-*.mk files when executed from depot

jschlatow, 2021-06-14:
Last week, I ran into a rather long debugging session when creating a bunch of api and src archives for existing library ports. In the end, I found an inconsistency between the normal use of the build system and its use from the depot tool. As I had to look into and understand the `base/mk/` files to narrow this down, I believe this inconsistency is quite hard to identify for the standard developer and deserves fixing.

When we build a binary that depends on a library, we add the library to the `LIBS` variable. Let's assume, we do `LIBS += foobar` in the corresponding `target.mk`. In consequence, the build system includes `lib/import/import-foobar.mk` which typically sets up the include paths as follows:

	INC_DIR += $(call select_from_ports,foobar)/include/foobar

Since I have seen several api archives that include the import files, I assumed the import files are also loaded when using the depot tool. In fact, this is true but the `select_from_ports` function behaves differently since there is no contrib dir in depot mode and consequently returns `REP_DIR`. This is reasonable as the `content.mk` of the api archive usually mirrors all required files from `PORT_DIR` and `REP_DIR` into a single directory that will be handled by the depot tool as a distinct repository. Yet, when the `import-foobar.mk` is included, `REP_DIR` will point to the importing repo not the imported repo. This is not much of a problem, if the archive adheres to the convention of putting include files into `include/` and `include/spec/{x86,x86_64,64bit,...}` since both, depot and Goa, set these include paths by default.

As a result, I got the following picture:
* import mk files are used by the build system in normal use
* import mk files are used by the depot tool but not 100% effective
* import mk files are not used by Goa

I also noticed that `import-libc.mk` uses a conditional to set `INC_DIR` when imported for the normal use of the build system and `REP_INC_DIR` when used by depot as a (partial) workaround. (In Goa, there is a hardcoded quirk for libc.)

However, I have found neither any documentation nor any clarifying output of the tools. So, how can we improve this situation? Here are a few suggestions and options on which I'd like to have your feedback:

Intuitively, I thought about correcting `REP_DIR` so that a call of `select_from_ports,foobar` would resolve to the directory of the archive in which the import mk file resides. On a second thought, however, include paths typically differ: In the contrib dir, the includes are typically found in `include/foobar` whereas in the api archive, these would be in `include`.

Another approach is to avoid the specification of any special `INC_DIR`s in import mk files and rather follow the aforementioned conventions more strictly. I could imagine a `$(call default_port_dirs,foobar)` that sets up the port-specific default `INC_DIR`s (including spec directories). This would expose when a library's import file implements any special treatment (and therefore requires special treatment in depot/Goa) or adheres to the standard scheme. For api archives, it would be helpful to clearly document the expected structure of the include files.

For the very least, I'd suggest printing a warning if an import mk file specifies a non-existing directory in `INC_DIR` to point the user to a potential misconfiguration. This would probably have the best cost-benefit ratio.

jschlatow, 2021-06-16:
Commits fb924c3 and 42245ca illustrate my suggestion of warning for missing include directories and the `default_port_includes` helper.
Commit e2135f3 demonstrates how this simplifies the import mk of the ffi library.

chelmuth, 2021-06-18:
I read your comments in this issues just now after nagging about ffi in #4201. I understand your suggest fix as prerequisite for ffi/glib.

jschlatow, 2021-06-18:
@chelmuth that's correct. I forgot to add the reference to #4201. I'm sorry for that.

I could split e2135f3 into two parts though to decouple both issues: one part relevant for both issues and one part only relevant for this one. Let me know if you prefer that.

chelmuth, 2021-06-18:
At the moment, I'd prefer a fixup for the current build issues with ffi and address this issue later.