genodelabs/genode #4194: falkon: search engines not stored persistently

jschlatow, 2021-06-14:
Falkon browser allows the customisation of search engines. Even though all other settings are stored persistently in my recall_fs, the search engines are reset once I restart falkon.

@cproc, can you reproduce this? Are you aware of any other (unsupported) storage location used by falkon?

cproc, 2021-06-21:
Fixed by 0e4dd1d. I've also uploaded an updated Sculpt 21.03 Falkon package. In addition to the fix it appears to be necessary to quit the browser in a 'clean' way (not killing or restarting from the Leitzentrale) to get changed settings stored in the sqlite3 database ("browsedata.db").

jschlatow, 2021-06-22:
@cproc that's great. Thanks for the quick investigation.

commit 17b3ab71 (2021-06-28): qt5: ignore 'fcntl(F_SETLK)' error in sqlite3 implementation

Fixes #4194
commit 3824c0ca (2021-07-28): qt5: ignore 'fcntl(F_SETLK)' error in sqlite3 implementation

Fixes #4194
closed 2021-07-28 by chelmuth