genodelabs/genode #4195: Qt5 window stacking problem

cproc, 2021-06-14:
The `qt5_stacking.run` test in commit [1] demonstrates a problem which occured when using the Falkon web browser. The main window contains a big push button. When the button gets clicked for the first time, a message window appears in front of the main window. But when the button gets clicked the next time (after closing the message window), the new message window appears behind the main window.

@nfeske: do you perhaps see anything suspicious with these logged reports which occurred during the interaction (I removed hover and pointer reports)?

1) before the button got clicked for the first time:

```
[init -> wm_report_rom] report 'layouter -> window_layout'
[init -> wm_report_rom]   <window_layout>
[init -> wm_report_rom]   	<window id="1" title="test-stacking -> Untitled Window" xpos="150" ypos="130" width="400" height="400" focused="yes" maximizer="yes" closer="yes"/>
[init -> wm_report_rom]   </window_layout>

[init -> wm_report_rom] report 'nitpicker -> focus'
[init -> wm_report_rom]   <focus label="wm -> test-stacking -> Untitled Window" color="#ffffff" domain="default" active="yes"/>

[init -> wm_report_rom] report 'layouter -> window_layout'
[init -> wm_report_rom]   <window_layout>
[init -> wm_report_rom]   	<window id="1" title="test-stacking -> Untitled Window" xpos="150" ypos="130" width="400" height="400" focused="yes" maximizer="yes" closer="yes">
[init -> wm_report_rom]   		<highlight>
[init -> wm_report_rom]   			< pressed="yes"/>
[init -> wm_report_rom]   		</highlight>
[init -> wm_report_rom]   	</window>
[init -> wm_report_rom]   </window_layout>
```

2) when the button got clicked for the first time and the message window appeared in front of the main window

```
[init -> wm_report_rom] report 'wm -> window_list'
[init -> wm_report_rom]   <window_list>
[init -> wm_report_rom]   	<window id="2" label="test-stacking -> Untitled Window.2" title="" width="98" height="69" resizeable="yes"/>
[init -> wm_report_rom]   	<window id="1" label="test-stacking -> Untitled Window" title="" width="400" height="400" resizeable="yes"/>
[init -> wm_report_rom]   </window_list>

[init -> wm_report_rom] report 'layouter -> window_layout'
[init -> wm_report_rom]   <window_layout>
[init -> wm_report_rom]   	<window id="1" title="test-stacking -> Untitled Window" xpos="150" ypos="130" width="400" height="400" focused="yes" maximizer="yes" closer="yes">
[init -> wm_report_rom]   		<highlight>
[init -> wm_report_rom]   			< pressed="yes"/>
[init -> wm_report_rom]   		</highlight>
[init -> wm_report_rom]   	</window>
[init -> wm_report_rom]   	<window id="2" title="test-stacking -> Untitled Window.2" xpos="300" ypos="230" width="98" height="69" maximizer="yes" closer="yes"/>
[init -> wm_report_rom]   </window_layout>

[init -> wm_report_rom] report 'layout_rules'
[init -> wm_report_rom]   <rules>
[init -> wm_report_rom]   	<screen name="screen"/>
[init -> wm_report_rom]   
[init -> wm_report_rom]   	<assign label="test-stacking -> Untitled Window.2" target="screen" xpos="300" ypos="230" width="98" height="69"/>
[init -> wm_report_rom]   	<assign label="test-stacking -> Untitled Window" target="screen" xpos="150" ypos="130" width="400" height="400"/>
[init -> wm_report_rom]   	<assign label_prefix="" target="screen" xpos="any" ypos="any"/>
[init -> wm_report_rom]   </rules>

[init -> wm_report_rom] report 'layouter -> window_layout'
[init -> wm_report_rom]   <window_layout>
[init -> wm_report_rom]   	<window id="2" title="test-stacking -> Untitled Window.2" xpos="300" ypos="230" width="98" height="69" maximizer="yes" closer="yes"/>
[init -> wm_report_rom]   	<window id="1" title="test-stacking -> Untitled Window" xpos="150" ypos="130" width="400" height="400" focused="yes" maximizer="yes" closer="yes">
[init -> wm_report_rom]   		<highlight>
[init -> wm_report_rom]   			< pressed="yes"/>
[init -> wm_report_rom]   		</highlight>
[init -> wm_report_rom]   	</window>
[init -> wm_report_rom]   </window_layout>

[init -> wm_report_rom] report 'layout_rules'
[init -> wm_report_rom]   <rules>
[init -> wm_report_rom]   	<screen name="screen"/>
[init -> wm_report_rom]   
[init -> wm_report_rom]   	<assign label="test-stacking -> Untitled Window.2" target="screen" xpos="300" ypos="230" width="98" height="69"/>
[init -> wm_report_rom]   	<assign label="test-stacking -> Untitled Window" target="screen" xpos="150" ypos="130" width="400" height="400"/>
[init -> wm_report_rom]   	<assign label_prefix="" target="screen" xpos="any" ypos="any"/>
[... 189 lines cut ...]
[init -> wm_report_rom]   </window_layout>

[init -> wm_report_rom] report 'layout_rules'
[init -> wm_report_rom]   <rules>
[init -> wm_report_rom]   	<screen name="screen"/>
[init -> wm_report_rom]   
[init -> wm_report_rom]   	<assign label="test-stacking -> Untitled Window" target="screen" xpos="150" ypos="130" width="400" height="400"/>
[init -> wm_report_rom]   	<assign label="test-stacking -> Untitled Window.2" target="screen" xpos="300" ypos="230" width="98" height="69"/>
[init -> wm_report_rom]   	<assign label_prefix="" target="screen" xpos="any" ypos="any"/>
[init -> wm_report_rom]   </rules>

[init -> wm_report_rom] report 'layouter -> window_layout'
[init -> wm_report_rom]   <window_layout>
[init -> wm_report_rom]   	<window id="1" title="test-stacking -> Untitled Window" xpos="150" ypos="130" width="400" height="400" focused="yes" maximizer="yes" closer="yes"/>
[init -> wm_report_rom]   	<window id="3" title="test-stacking -> Untitled Window.2" xpos="300" ypos="230" width="98" height="69" maximizer="yes" closer="yes"/>
[init -> wm_report_rom]   </window_layout>

[init -> wm_report_rom] report 'layouter -> window_layout'
[init -> wm_report_rom]   <window_layout>
[init -> wm_report_rom]   	<window id="1" title="test-stacking -> Untitled Window" xpos="150" ypos="130" width="400" height="400" focused="yes" maximizer="yes" closer="yes"/>
[init -> wm_report_rom]   	<window id="3" title="test-stacking -> Untitled Window.2" xpos="300" ypos="230" width="98" height="69" maximizer="yes" closer="yes"/>
[init -> wm_report_rom]   </window_layout>
```

[1] https://github.com/cproc/genode/commit/748c15ce8b90d709d2f7afd06a013c2021040fb6

nfeske, 2021-06-14:
Thanks for documenting the issue so well. I think I have an explanation.

Whenever a window is raised, the window layouter puts the window's `<assign>` rule to the top. This is intuitively fine.

Whenever a window appears for which no concrete `<assign label=...>` rule exists and an `<assign label_prefix=...>` wildcard rule matches, a new `<assign label=...>` rule is put in front automatically. This is intuitively fine, too. This case covers the initial appearance of the alert window.

As soon as the alert window appeared for the first time, a concrete `<assign label="...Window.2"/>` rule exists. When the alert window is closed, the rule stays in tact. Now, when the main window gets activated, the main window's `<assign>` rule will be moved above all other `<assign>` rules, including the rule for the (currently absent) alert window.

When the alert window appears for the next time, the window layouter finds a matching concrete `<assign>` rule. Since the rule is below the main window's `<assign>` rule, the alert window is deliberately placed behind the main window.

In other words: Everything works as designed. It merely doesn't work as intended. ;-)

What would be the good approach to fix it?

It think that the problem lies in the overzealous approach of bringing the `<assign>` rule above of all other windows whenever a window is raised. It works well in relation to all currently present windows but it fails to recognize the meaning of the assign rules for currently absent windows. If a currently absent window had been in front of the raised window, it should stay in front of it. So when it re-appears, it will be displayed top-most.

Right now, the window layouter does not make this distinction between present and non-present windows (aka used or unused assign rules). See https://github.com/genodelabs/genode/blob/master/repos/gems/src/app/window_layouter/main.cc#L504 for the code. The current implementation creates the sorting order by calling `_gen_rules_assignments` twice. Once for the most recently raised window, and once for all others. A better solution would be to call it three times. Once for all absent windows that appear in front of the most recently raised window, once for the raised window, and once for all others. I haven't looked whether we already have an easy way to distinguish used `assign` entries from unused ones though.

@cproc Would you like to try implementing this or would that distract you too much from more important/interesting work? Should it would take you more than - let's say - a day, I'd offer to step in.

cproc, 2021-06-15:
The problem could also be in Qt, or at least a part of it From Qt's perspective, a new message window is created each time the button is clicked and it should always appear on top, but currently the label/title of the previous (deleted) message window is reused and so the window layouter appears to believe that the previous window reappeared. If it actually were the previous message window, it would be correct to put it in the back, because the focus went to the main window when the button got clicked again. The test works correctly when I show and hide the same window controlled by a timer. It also works correctly when I add a new ID to the label/title of each newly created Qt window, but I noticed that then the 'layout_rules' report fills up with all the previous labels of destroyed windows. Is it supposed to be this way that window labels are remembered forever or is something missing to remove a rule when a window gets destroyed?

cproc, 2021-06-15:
Commit 58dd3e1 contains the mentioned changes to the test. Currently, the timer-controlled show and hide of the same window is enabled. With this setup, I noticed another peculiarity: when the small window overlaps with the main window and the small window is on top: when I double-click on a part of the small window which is outside of the main window, next time the small window is shown, it sometimes (not always) appears behind the main window.

cproc, 2021-06-15:
I mean partially overlap: a part of the small window overlaps with the main window and another part is outside the main window.

cproc, 2021-06-15:
Commit 3ac6429 has the window title ID update added to the test as well (I had it implemented in the QPA plugin before, but it is easier to switch on or off in the test application).

nfeske, 2021-06-17:
Thank you for the further testing. I'll look into this issue in more detail next week and implement my idea for the window layouter.

>  Is it supposed to be this way that window labels are remembered forever or is something missing to remove a rule when a window gets destroyed?

That is the intention behind the rules database. Hence, the approach of generating new labels for each repeated ocurrence of the same window (filling up the rules database) works against the grain of the concept. Note that in a scenario where the rules are made persistent (like in my Sculpt installation where I store them in my recall fs), the unique labels would be kept present across reboots or application restarts. So if the restarted application generates the "unique" IDs in a deterministic way, the `<assign>` rules of the previous run are applied.

It would be best to have one unique label for each window and reuse the label whenever the window re-appears. I'm hopeful that the suggestion I gave above works out. Otherwise, we'd have to think about a way to distinguish different kinds of windows (temporary alert windows, regular application windows) and apply the rules database only actual application windows but I'd prefer to avoid introducing such a special-case handling.

nfeske, 2021-06-23:
I was able to reproduce the problem by opening, closing, and reopening the about-windows in the qt5_textedit scenario.

Commit https://github.com/nfeske/genode/commit/dae2c3e21e4ca928229d16e461a39a9b769b129b implements the idea outlined above. It fixes the stacking-order problem for the textedit scenario. @cproc Could you give it a try in the Falkon-Szenario?

nfeske, 2021-06-23:
@cproc I just noticed that it's too early for testing the patch. I keep working on it.

nfeske, 2021-06-23:
Here is my next take: https://github.com/nfeske/genode/commit/923950c304bfd5a71027e8a9d1b0018a0067bd95 tested with both Sculpt and qt5_textedit.run. I'd appreciate your feedback, @cproc.

cproc, 2021-06-24:
@nfeske: I can confirm that your solution fixes the stacking issues with Falkon dialogs which have been observed so far. It does not solve all issues simulated by the test application (if a test window was put into the background by the user clicking on the main window and the test window gets destroyed by the application while it was in the background and a new test window with the same title gets created, the new test window appears in the background), but we can reopen this issue if this special case would ever be a problem in practice.

nfeske, 2021-06-24:
I'm afraid that I'm out of ideas how the latter case can be solved in principle because it contradicts with the window layouter's approach to restore the window layout from a previous boot. If the user has explicitly put a window in the back, it re-appears in the back. That is intentional.

cproc, 2021-06-24:
How about if the user could choose at the start of an application if the window layout rules for this application should be remembered (with the possible side-effects) or not at all? Is this something which could already be achieved with session routing rules?

cproc, 2021-06-24:
(Only theoretically speaking, maybe this won't ever be necessary in practice)

cproc, 2021-06-24:
If the layout rules were not remembered at all, a window which got hidden and shown again (versus destroyed and newly created) would always appear in front, which would also not be correct. So, this would also not be a complete solution.

nfeske, 2021-08-31:
@cproc do you agree to close this issue? I don't foresee any further improvements.

closed 2021-08-31 by cproc