genodelabs/genode #4198: vfs/dir_fs: allow opening empty dirs

m-stein, 2021-06-15:
The <dir> file system returns a "lookup failed" error if one attempts to open a path to a directory that has no sub-file-system. This behavior is at least unintuitive and I couldn't find a technical reason for it. An example would be using the FS-query tool with the following configuration:
```
<vfs>
  <dir name="x">
    <dir name="y"/>
  </dir>
</vfs>
<query path="/x/y" />
```
The tool will not report an empty directory (the report will be empty instead) because of the error return value when ``Dir_file_system::open_composite_dirs`` reaches ``y``. By the way, when having an empty root directory, the problem is the same:
```
<vfs/>
<query path="/" />
```
Only that this time, FS-query doesn't catch the exception caused by the error and therefore aborts:
```
[init -> fs_query] Error: Uncaught exception of type 'Genode::Directory::Nonexistent_directory'
```

m-stein, 2021-06-15:
Commit c9c262f498 fixes the issue and I could run the following run scripts successfully with it (nova/pc):
```
fs_query
vfs_cfg
vfs_import
libc_vfs_ext2
libc_vfs_fs_ext2
```

While testing, I noticed that libc_vfs_component.run contains a never called ``run_genode_until``:
```
run_genode_until forever                                                        
run_genode_until "child \"test-libc_counter-sink\" exited with exit value 0.*\n" 30
```
... that run/vfs_stress_rump_fs fails on staging:
```
[init -> vfs_stress] generating directory surface...
[init -> vfs_stress] created 3162 empty directories, 21397μs/op , 604KiB consumed
[init -> vfs_stress] generating files...
[init -> vfs] rump: uid 0 on /: out of inodes
[init -> vfs_stress] Error: OPEN_ERR_NO_SPACE
[init -> vfs_stress] Error: failed at '/1/a/a/a/a/b/b/b/b/b/b/b/b/b/c' after 390 files
[init -> vfs_stress] Error: Uncaught exception of type 'Genode::Exception'
[init -> vfs_stress] Warning: abort called - thread: ep
```
... and run/vfs_stress_ext2 as well:
```
[init -> vfs_stress] generating directory surface...
[init -> vfs_stress] created 210 empty directories, 6338μs/op , 5092KiB consumed
[init -> vfs_stress] generating files...
[init -> vfs_stress] created 210 empty files, 3761μs/op, 8164KiB consumed
[init -> vfs_stress] writing files...
[init -> vfs_stress] wrote 2646 bytes, 2kB/s, 8164KiB consumed
[init -> vfs_stress] reading files...
[init -> vfs_stress] read 2646 bytes, 6kB/s, 8164KiB consumed
[init -> vfs_stress] unlink files...
[init -> vfs_stress] recursive unlink not supported
[init -> vfs_stress] Error: OPENDIR_ERR_LOOKUP_FAILED
[init -> vfs_stress] Error: unlink /0 failed
[init -> vfs_stress] Error: Uncaught exception of type 'Genode::Exception'
[init -> vfs_stress] Warning: abort called - thread: ep
```

nfeske, 2021-06-17:
Thank your for addressing this issue with so much care. The extension of the test and the explanations given in the comments are very much appreciated. I tried to cherry-pick your commit to staging but it does not cleanly apply. Could you rebase the commit to the current staging branch?

Thank you also for revealing the broken run scripts (apparently not exercised by our autopilot). I'm admittedly not quite sure how to go about them.

m-stein, 2021-06-17:
There was another FS-query path in the pipe:
34087ca18a vfs/dir_fs: allow opening empty dirs
ae6440a9dd fs_query: gracefully deal with missing directories

nfeske, 2021-06-17:
Thanks for clarifying. I merged both commits along with the additional fixup bceb240304d12fc737c38999456b9b813969eae5.

m-stein, 2021-06-18:
Oh, I'm sorry for the inconvenience! I'll check it better next time.

commit 5528c285 (2021-06-17): vfs/dir_fs: allow opening empty dirs

Adapts Dir_file_system::open_composite_dirs in a way that it returns "success"
when the leaf node of the path is an empty directory but "lookup failed", as
usual, if one of the other directories on the way to the leaf node is empty.

I couldn't find a technical reason why we used to return "lookup failed" when
only the leaf node was empty.

The commit also adds a test for en empty root directory and empty
sub-directories to the fs_query run script.

Fixes #4198
closed 2021-06-25 by chelmuth