genodelabs/genode #4199: tool/dts/extract: does not extract the whole transitive tree

skalk, 2021-06-16:
When using the new `extract` tool on a preprocessed MNT Reform2 dts file to extract the Display Controller Subsystem (DCSS), via:
```
tool/dts/extract --extract dcss
``` 
not all nodes of the transitive closure of referenced nodes get extracted.

skalk, 2021-06-16:
Commit af5ebd5 showcases the problem.

nfeske, 2021-06-22:
Thanks @skalk for bringing up the limitation of the extract tool.

Upon closer inspection, the problem lies in the way of how transitive dependencies are selected.

If a node is explicitly selected (via a --select argument) or directly referenced by another node, it is preserved with all its information. However, if a node does merely happen to be a parent node of a selected node, only the node's properties are preserved but the sibling nodes of the selected node are deliberately cut off. When following a selected node towards the root of the device tree, all properties along the way are preserved, but (presumably unrelated) sibling nodes are not. The cutting off of sibling nodes is important. Otherwise, if we would capture all sibling nodes also, we would pull in all SoC parts once we reach the /soc node, which is obviously not what we want.

Now, the problem with this heuristic is that some properties are disguised as nodes, specifically cross-references specified in so-called `ports` nodes. Strictly speaking, those kind of nodes need to be preserved. But they aren't because the extract tool sees them merely as sibling nodes of a selected node and discards them.

We could in principle refine the heuristics of the extract tool, e.g., preserve a node completely if it is named "ports". However, I'm not quite sure how far this would take us. The ports-node seems to be a rather informal convention. There may exist other patterns to a similar effect.

Hence, for now, I'd not try to make the tool more watertight before we have gained a good grasp on the patterns used in dts files across a variety of SoCs. Until then, on a case-by-case basis, the user might need to specify multiple --select arguments and cross-check the proper functioning of the extracted device tree by feeding it to Linux.

closed 2021-06-22 by nfeske