genodelabs/genode #4200: Uncaught exception in nic_router

jschlatow, 2021-06-16:
I just triggered an uncaught exception in nic_router when copying the nic_router.config generated by mbimcli to `/config/nic_router`:

```
[runtime -> nic_router] [uplink] static IP config: interface 10.113.59.159/16, gateway 10.113.59.1 P2P 0
[runtime -> nic_router] [uplink] NIC sessions: 0
[runtime -> nic_router] [default] static IP config: interface 10.0.3.1/24, gateway 0.0.0.0 P2P 0
[runtime -> nic_router] [default] NIC sessions: 0
[runtime -> nic_router] [default] NIC sessions: 1
[runtime -> nic_router] [uplink] NIC sessions: 1
[runtime -> nic_router] [default] NIC sessions: 1
[drivers -> platform_drv] 00000000:00000014.3 power off
[runtime -> nic_router] [uplink] dynamic IP config: none
[runtime -> nic_router] [uplink] NIC sessions: 0
[runtime -> nic_router] [default] NIC sessions: 2
[runtime -> nic_router] Error: Uncaught exception of type 'Genode::Bit_array_dynamic::Invalid_index_access'
[runtime -> nic_router] Warning: abort called - thread: ep
```

Before copying the file, the nic_router was running in managed mode, i.e. `/config/nic_router` was not present. It was uplinked via wifi_drv. I performed the same steps a couple of times before without any error, hence this is not easily reproducible. Note, that the power off message from platform_drv belongs to the wifi controller.

This is the nic_router.config I copied:
```
<config verbose="no" verbose_packets="no" verbose_domain_state="yes">
        <default-policy domain="default"/>
        <policy label_prefix="usb_modem_drv" domain="uplink"/>
        <domain name="uplink" interface="10.113.59.259/16" gateway="10.113.59.1" use_arp="no">
                <nat domain="default" tcp-ports="100" udp-ports="100" icmp-ids="100"/>
        </domain>
        <domain name="default" interface="10.0.3.1/24">
                <dhcp-server ip_first="10.0.3.2" ip_last="10.0.3.200">
                        <dns-server ip="62.109.121.17"/>
                        <dns-server ip="62.109.121.18"/>
                </dhcp-server>
                <tcp dst="0.0.0.0/0">
                        <permit-any domain="uplink"/>
                </tcp>
                <udp dst="0.0.0.0/0">
                        <permit-any domain="uplink"/>
                </udp>
                <icmp dst="0.0.0.0/0" domain="uplink"/>
        </domain>
</config>
```

What puzzles me is that the nic_router first correctly logs a single session in the uplink and default domain but then switches to 0 sessions in the uplink domain and 2 sessions in the default domain.

m-stein, 2021-06-17:
@jschlatow Thanks for reporting this issue in such detail! One idea I got from looking at your description is that the number of ports/ids configured in the NAT rule of your uplink is rather small compared to the default in Sculpt (udp-ports="1000" tcp-ports="1000" icmp-ids="1000"). However, this doesn't explain why your session switches to ``default``. Maybe this has something to do with the default policy and the uplink domain disappearing for some reason. I'll try to reproduce the problem in a minimal scenario.

m-stein, 2021-06-17:
I will continue testing once I'm home as the test I've created troubles the office LAN (the uplink session might connect to the default domain with a DHCP server that interferes with our local server).

jschlatow, 2021-06-22:
@m-stein it seems that I can reproduce this issue consistently with my setup. That means it occured both times I tested the usb_modem_drv yesterday and this morning. Let me know if you need more details.

m-stein, 2021-06-22:
@jschlatow: Could you give c04892d899 and 77f0e627b7 a try and send me the log with the uncaught exception?

jschlatow, 2021-06-23:
@m-stein, I attached the [log](https://github.com/genodelabs/genode/files/6699696/nic_router_log.txt). It shows two "invalid index while freeing" messages.

chelmuth, 2021-06-23:
Hm, the NIC router obviously tries to free `10.0.1.3`  in `IP range: first 10.0.3.2 last 10.0.3.200` which is at negative offset `4294966785 (0xfffffe01)`.

m-stein, 2021-06-24:
@jschlatow Thanks for sending the log. It seems that the problem comes from the fact that you're re-configuring domain default without changing the name but changing the DHCP subnet from 10.0.1.x to 10.0.3.x. I think with this, I can write a minimal test case.

m-stein, 2021-06-24:
Commit 3932d782cbaebf6a10b493a5072d56d6898a4d1f fixes the issue and adds a test.

@jschlatow Could you give it a try?

chelmuth, 2021-06-25:
Do I understand correctly that the patch addresses the DHCP-domain handling on a upper layer only? While it is good to fulfill the contract and use sane parameters for `free_ip` only, but the use of the bit allocator is still not wrapped by try-catch clauses (as in the do-not-merge commits) and may hit us again.

m-stein, 2021-06-25:
@chelmuth Of course, we can guard also against the exceptions although they should never occur.

m-stein, 2021-06-25:
Commit 41a7bdd5d3 adds the guard. However, I have added it only in ``Dhcp_server::free_ip`` and not in ``Bit_allocator_dynamic::Bit_allocator_dynamic`` resp. ``Bit_array_dynamic::reserve`` as the exception there is prevented from the logic in ``Bit_allocator_dynamic::Bit_allocator_dynamic``.

m-stein, 2021-06-25:
Sorry, meant this one: ef68ad9400.

jschlatow, 2021-06-25:
@m-stein, I tested 3932d78 and can confirm that it solves the issue.

m-stein, 2021-06-25:
@jschlatow Thanks for testing it ;)

commit be644098 (2021-06-25): nic_router: fix exc. in Interface::handle_config_3

When Interface::handle_config_3 (third step of applying a new configuration to
interfaces) tried to detach the interface from the current IP config because
the old and new IP config differed, it did so using the new domain. The former
steps of the reconfiguration already installed the new domain reference at the
interface. Therefore, also the DHCP server of the new domain was used. This,
however caused uncaught exceptions because detaching from an IP config
includes dissolving all DHCP allocations. This dissolving of DHCP allocations
now oper [... cut]
closed 2021-07-28 by chelmuth