genodelabs/genode #4201: depot: add archives for usb_modem_drv and ffi library

jschlatow, 2021-06-16:


jschlatow, 2021-06-16:
Comments:

* ffi library is required for mbimcli (in genode-world)
* When testing the usb_modem_drv on my t490s with a speedtest, it failed in `netif_stop_queue()` due to a `TRACE_AND_STOP`. Commit 00fb1f2 removes the `TRACE_AND_STOP` (other drivers don't implement the function either). So far, the usb_modem_drv runs fine with this change.

chelmuth, 2021-06-18:
Unfortunately, the ffi archives do not build.
```
created genodelabs/api/ffi/2021-06-18-a
created genodelabs/api/libc/2021-05-04
created genodelabs/src/ffi/2021-06-18-a
checking library dependencies...
  Library ldso_so_support
    COMPILE  so_support.o
    MERGE    ldso_so_support.lib.a
  Library libc
    CONVERT  libc.symbols.s
    ASSEMBLE libc.symbols.o
    MERGE    libc.abi.so
  Library ffi
    COMPILE  ffi64.o
.../genode/depot/genodelabs/src/ffi/2021-06-18-a/src/lib/ffi/src/x86/ffi64.c:30:10: fatal error: ffi.h: No such file or directory
   30 | #include <ffi.h>
      |          ^~~~~~~
compilation terminated.
.../genode/repos/base/mk/generic.mk:52: recipe for target 'ffi64.o' failed
make[6]: *** [ffi64.o] Error 1
var/libdeps:60: recipe for target 'ffi.lib' failed
make[5]: *** [ffi.lib] Error 2
Makefile:301: recipe for target 'gen_deps_and_build_targets' failed
make[4]: *** [gen_deps_and_build_targets] Error 2
.../genode/tool/depot/mk/build_bin_archive:201: recipe for target '.../genode/depot/genodelabs/bin/x86_64/ffi/2021-06-18-a.build/bin' failed
make[3]: *** [.../genode/depot/genodelabs/bin/x86_64/ffi/2021-06-18-a.build/bin] Error 1
.../genode/depot/var/build.mk:12: recipe for target 'genodelabs/bin/x86_64/ffi/2021-06-18-a' failed
make[2]: *** [genodelabs/bin/x86_64/ffi/2021-06-18-a] Error 2
.../genode/tool/depot/build:134: recipe for target 'execute_generated_build_mk_file' failed
make[1]: *** [execute_generated_build_mk_file] Error 2
tool/depot/create:59: recipe for target 'build' failed
make: *** [build] Error 2
```

Also glib (from world) does not build due to a similar error. The reason is that `select_from_ports` in the .mk files does not pick up *ffi.h* from the api package.

chelmuth, 2021-06-18:
Also, there's no x86_32 version of ffi. @ssumpf did you omit it by intention?

jschlatow, 2021-06-18:
@chelmuth, please consider f0bf9fb as a fixup for the build issues.

chelmuth, 2021-06-18:
Great, merged to staging.

ssumpf, 2021-06-23:
> Also, there's no x86_32 version of ffi. @ssumpf did you omit it by intention?

ffi was ported on demand (x86_64, arm_v7 for Java and arm_v8 for glib because of libmbim and i.MX8 modem support). It should not be hard to enable it for x86_32. @jschlatow: Shall I look into it, or do you want to do it?

ssumpf, 2021-06-23:
@jschlatow: Never mind, I am almost done with the x86_32 version.

jschlatow, 2021-06-24:
@ssumpf that's very nice to hear. Sorry for the late reply though.

ssumpf, 2021-06-24:
@chelmuth: 3b58110 adds x86_32 support in `ffi`. @jschlatow: Could you add the required things to the ffi recipes?

jschlatow, 2021-06-24:
Sure, I just pushed 9743308, which adapts the ffi recipe.

chelmuth, 2021-06-24:
Thanks @ssumpf and @jschlatow! I just merged the commits to staging.

closed 2021-06-25 by chelmuth