genodelabs/genode #4202: wifi_drv: compiler warnings since GCC 10

m-stein, 2021-06-17:
Compiling the wifi driver gives me many compiler warning of this type:
```
repos/dde_linux/src/lib/wifi/include/lx_emul.h: In function ‘page_ref_inc’:
repos/dde_linux/src/lib/wifi/include/lx_emul.h:365:13: warning: taking address of packed member of ‘struct page’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  365 |  atomic_inc(&page->_count);
      |
```

cnuke, 2021-06-17:
@m-stein please try removing the `packed` attribute from the struct definition.

m-stein, 2021-06-17:
I'm not sure if it's a good idea to just remove the ``packed`` attribute, but in an offline discussion with @cnuke we settled on giving it a try. If I remove the packed attribute from ``struct page`` all warnings in ``lx_emul.h`` disappear. However, there are some left in other files:
```
repos/dde_linux/src/lib/wifi/dummies.cc:213:10: warning: declaration of ‘DUMMY isalpha()’ conflicts with built-in declaration ‘int isalpha(int)’ [-Wbuiltin-declaration-mismatch]
repos/dde_linux/src/lib/wifi/dummies.cc:285:10: warning: declaration of ‘DUMMY sscanf()’ conflicts with built-in declaration ‘int sscanf(const char*, const char*, ...)’ [-Wbuiltin-declaration-mismatch]
repos/dde_linux/src/lib/wifi/dummies.cc:461:10: warning: declaration of ‘DUMMY strncpy()’ conflicts with built-in declaration ‘char* strncpy(char*, const char*, long unsigned int)’ [-Wbuiltin-declaration-mismatch]
repos/dde_linux/src/include/lx_emul/impl/pci.h:47:23: warning: comparison of integer expressions of different signedness: ‘const __u32’ {aka ‘const unsigned int’} and ‘int’ [-Wsign-compare]
repos/dde_linux/src/lib/wifi/lxcc_emul.cc:1554:17: warning: invalid conversion from ‘void*’ to ‘addr_t’ {aka ‘long unsigned int’} [-fpermissive]
repos/dde_linux/src/lib/libnl/socket.cc:253:28: warning: ‘w_msg.Wifi::Msghdr::msg_count’ may be used uninitialized in this function [-Wmaybe-uninitialized]
repos/dde_linux/src/lib/libnl/socket.cc:312:28: warning: ‘w_msg.Wifi::Msghdr::msg_count’ may be used uninitialized in this function [-Wmaybe-uninitialized]
repos/dde_linux/src/lib/wpa_driver_nl80211/dummies.cc:57:10: warning: declaration of ‘DUMMY rint()’ conflicts with built-in declaration ‘double rint(double)’ [-Wbuiltin-declaration-mismatch]
repos/dde_linux/src/lib/wpa_supplicant/ctrl_iface_genode.c:227:27: warning: taking address of packed member of ‘struct Msg_buffer’ may result in an unaligned pointer value [-Waddress-of-packed-member]
repos/dde_linux/src/lib/wpa_supplicant/ctrl_iface_genode.c:228:27: warning: taking address of packed member of ‘struct Msg_buffer’ may result in an unaligned pointer value [-Waddress-of-packed-member]
repos/dde_linux/src/lib/wpa_supplicant/ctrl_iface_genode.c:232:28: warning: taking address of packed member of ‘struct Msg_buffer’ may result in an unaligned pointer value [-Waddress-of-packed-member]
```

m-stein, 2021-06-17:
My suggestion for a fix: 810d5d4911ca710f869e57c932fdc8a1da07fc52.

m-stein, 2021-06-17:
I tested 810d5d4 with run/nic_router_uplinks.

commit 69074e13 (2021-06-18): wifi_drv: fix compile warnings with gcc 10

Fixes #4202
closed 2021-06-25 by chelmuth