genodelabs/genode #4203: Bad network throughput in native falkon browser

jschlatow, 2021-06-17:
When test driving the usb_modem_drv, I ran a few web-based speedtests and stumbled upon a huge difference in throughput depending on whether the test is run in a VM-based falkon or in the Sculpt-native falkon.

The effect occurs for LTE (via usb_modem_drv) as well as VDSL (via LAN). In both cases, I got around 50Mbit/s downstream in my VM but only 10Mbit/s downstream in the native falkon. I tried different websites with similar results (some don't even finish due to the low throughput). During the test, the `falkon` thread loads a CPU core up to 50%.

jschlatow, 2021-12-17:
In order to narrow down the cause this issue, I ran some tests with netperf. More precisely, I was running netperf server on Sculpt 21.11 on a Thinkpad x260 and the client from within my Linux VM on my productive Sculpt 21.10 using vbox6. When using lxip, I got a throughput of 90Mbit/s (it's a 100Mbit/s Ethernet). However, with lwip, I merely achieved approx. 50Mbit/s. This effect was not reproducible, though, when reversing the direction of the TCP stream, using the _TCP_MAERTS_ test, or running the client on a native Linux.

I had more detailed look at it with various tracing mechanisms (including wireshark) and found that, in the lwip case, there are a bunch of pauses of ~200ms in which none of the involved components seems to make any progress. (More details of the story will be shared in a future genodians article.) Looking at wireshark traces, I also observed that there is a large number of retransmissions when the client is running in the VM (irrespective of the TCP/IP stack), but none when the client is running on a native Linux.

As it turns out, both stacks only acknowledge every other TCP segment on average (adhering to RFC5681). In some situations, when retransmissions take place, the sender (i.e. the netperf client) is waiting for an ACK of the receiver after retransmitting a segment. If this, however, coincides with the case that the receiver does not ACK the retransmitted segment and there are no more segments in flight, the sender will run in a timeout. This explains the observed pauses. Interestingly, lxip seems to deal with this corner case appropriately whereas lwip does not.

I haven't identified the exact condition yet, but confirmed that acknowledging every single packet in lwip increases the throughput to the expected 90Mbit/s. As a sidenote, I tried disabling the nagle algorithm but this did not make any noticeable difference. I'm planning to file this issue upstream once I'm able to extract pcap traces within Sculpt (see #4352). Such a mechanism would also help investigating what causes the retransmission in the first place.

Unfortunately, the missing ACKs are not the major bottleneck in the falkon case. I tried my patched vfs_lwip with falkon 2021-10-13 and got a downstream throughput of 17Mbit/s (vs. 16Mbit/s).

jschlatow, 2022-02-14:
For the record, I tried #4427 with falkon but do not see any improvement.

jschlatow, 2022-06-02:
I just re-run the speedtest on my x260 with Sculpt 22.04. Interestingly, the throughput depends on whether I use the ram_fs or the USB stick. The difference is significant (20Mbit/s from ram_fs vs 10Mbit/s from USB stick). I first suspected file system accesses (I'm using recall_fs for download and config) and let recall_fs use the ram_fs even when using the USB stick. However, I still get only 10Mbit/s in this setup.

@nfeske, @cproc: Do you have any idea what might be the culprit here?

nfeske, 2022-06-02:
Thanks @jschlatow for sharing your observations that I find quite surprising. Given that recall_fs (chroot) does not wrap the actual session interface but only intercept the session creation, I can only think of two possible effects.

1. The session arguments observed by the ram_fs might differ between both scenarios (with and without recall_fs in-between ram_fs and the browser). Would it be possible that the `tx_buf_size` argument differs for some reason? If the file-system session created via the recall_fs uses a smaller buffer size, file operations would become slower.

2. The file hierarchy at the ram_fs side is deeper when using the recall_fs. This might come into effect when many files are dynamically opened/closed because the hierarchy of parent directories much also be opened/closed for each file. The comparison of the traces of the file operations in the ram_fs in both scenarios would be insightful. With the new trace_logger support for Sculpt, I'm positive that such traces can be captured without too much effort.

nfeske, 2022-06-02:
Regarding the first possible effect, the session arguments arriving at ram_fs can be revealed in Sculpt by enabling the `<report provided="yes"/>` report in the /config/managed/runtime and looking at /report/runtime/state.

jschlatow, 2022-06-02:
I played a bit around and noticed that my first conclusion was probably a bit hasty. It appears that the bad throughput situation is independent of the used storage. Instead, it seems to be a side effect of restarting the intermediate chroot component while falkon is running. In approx. 50% of the cases I ended up with 5-10Mbit/s throughput. By restarting falkon (once or twice), I was able to recover and get 20MBit/s. This behaviour occurs independent of the used storage.

jschlatow, 2022-07-08:
I use this ticket to keep track of my recent findings. Unfortunately, I'm still quite far from identifying the culprit of the bad throughput for the falkon browser, however, I managed to dig up a few issues in our network infrastructure.

First of all, the NIC router has recently seen significant improvements mainly by avoiding exception catching (#4536) and timer RPC (#4555). This boost the UDP throughput but does not seem to have a large impact on TCP.

With vfs_lwip I currently get the following throughput with a x260 when sending/receiving UDP streams via a NIC router to a fast receiver/sender.
- TX: 680Mbit/s
- RX: 370Mbit/s (impacted by `Packet_alloc_failed` exceptions due to overload scenario)

My last measurements with netperf on an x260 showed the following numbers:
- TCP_STREAM lwip: 160Mbit/s
- TCP_MAERTS lwip: 135Mbit/s
- TCP_STREAM lxip: 230Mbit/s
- TCP_MAERTS lxip: 100Mbit/s

One aspect is that vfs_lwip does not batch ack_avail and packet_avail signals at all. This is more of an issue on base-hw than on base-nova, though. 

Another aspect is that there is a lot of computational overhead in the socket_fs_plugin. I did a bit of profiling with GENODE_LOG_TSC (numbers are for 500k calls):
- `do_sendto()`: 24G
  - `write(remote_fd)`: 6G (<- potential optimisation)
  - `write(data_fd)`: 17G
    - `pbuf_alloc()`: 2100M
    - `udp_sendto()`: 6700M
    - `pbuf_free()`: 1200M
    - remaining: 6G (<- potential optimisation)

Note that `do_sendto()` performs two write operations. The "overhead" appears to be around 12000 cycles on average for a write operation.

jschlatow, 2022-10-17:
On Sculpt 22.10, it seams the throughput has deteriorated a little. I made a few brief measurements. On the x260, I got 11Mbit/s whereas, on the T490s, I merely got 8Mbit/s. I haven't looked into this yet, however, given that my Linux VM only get's 22Mbit/s in the same test, I'm suspecting this might be a general degradation.

chelmuth, 2022-10-18:
I just tested my VM under Sculpt 22.10 and got the following result (which is Megabyte per second).

```
wget -O /dev/null http://URL/0512_MiB
...
2022-10-18 13:01:59 (20,3 MB/s) - ‘/dev/null’ saved [536870912/536870912]
```

jschlatow, 2022-10-20:
I believe my measurements from a few days ago have been impaired by some external factors. I retested my machines today and got the original values (~16Mbit/s with native Falkon, 50Mbit/s (limited by my DSL speed) with any browser in my Linux VM).

nfeske, 2023-04-26:
@jschlatow that's very nice to read!

I wonder, are you planning to further investigate the remaining difference? If not, could this issue be closed?

jschlatow, 2023-04-26:
@nfeske I'm not planning to investigate this in the near future as I'd rather focus on a few other tasks.

Regarding the performance improvement: I'm not sure how much of the improvement is caused by the upgrade of my DSL line from 50 to 100Mbit/s. I therefore just ran the same tests with mobile network and got 13Mbit/s for native falkon and 23Mbit/s in my Linux VM. It thus still seems to be unnecessarily slow, which is why I'd like to keep this issue open.

nfeske, 2023-04-26:
That makes perfect sense.

jschlatow, 2023-06-06:
I just took the opportunity to run a speedtest with a 300Mbit/s line from Deutsche Glasfaser. With native falkon, I get almost 100Mbit/s download speed and 110Mbit/s upload speed. In my Linux VM ist 180Mbit/s download and 116Mbit/s upload. These are interesting results because it seems that only the download speed is significantly different. This might be worth knowing for future investigation.

Note that I have acquired these results with wifi and while running native falkon and the Linux VM simultaneously, which slows down the download speed. I get 230Mbit/s when running the Linux VM solely.

chelmuth, 2023-06-06:
Note, while investigating #4821, I discovered that VirtualBox yields asymmetric download/upload bandwidth also in local scenarios. Today, I measured 115 MB/s upload and 60 MB/s download throughput with dd-over-ssh and the upcoming pc_nic_drv.

jschlatow, 2023-10-16:
Today, I ran bandwidth tests (using breitbandmessung.de) with Sculpt 23.10 RC on a 100Mbit VDSL line and got the following results (averaged over three runs each).

| Browser | Latency | Downstream | Upstream |
| --- | --- | --- | --- |
| Falkon in VM | 12 ms | 63 Mbit/s | 36 Mbit/s |
| Falkon | 15 ms | 30 Mbit/s | 27 Mbit/s |
| Falkon-jemalloc | 14 ms | ~ 60Mbit/s (*) | n/a |
| Morph | 15ms | 14 Mbit/s | 30Mbit/s |
| Morph-jemalloc | 14ms | ~54Mbit/s (*) | n/a |

(*) I was unable to complete the tests with the jemalloc variants since these failed with illegal READs at some point during the upstream or downstream test.

cproc, 2023-10-20:
I got `malloc()`-related errors with the non-jemalloc browser variants during the `breitbandmessung.de` test and found the cause in parallel executions of `vsnprintf()`, which was not thread-safe. I published new Sculpt 23.10 browser packages using the latest libc thread-safety commits from issue #725 and the bandwidth tests succeeded for me with all browser variants, although I had to increase the browser RAM quota in my local depot because of resource requests (even before the libc changes).

closed 2025-04-23 by jschlatow

jschlatow, 2025-04-23:
By the way, I'm getting the same results for morph_browser-jemalloc/2025-04-11.