genodelabs/genode #4208: qt5: implement 'SyncSocket' with pipes on Genode

cproc, 2021-06-24:
QtWebEngine uses `SyncSocket` objects to synchronize audio streaming. These objects are implemented with socket pairs of type `AF_UNIX` on POSIX platforms, which are currently not available on Genode. It appears to be possible to implement `SyncSocket` with pipes instead on Genode (we won't know for sure until a real audio backend becomes functional, but at least this implementation already makes it possible to play some muted YouTube videos).

cproc, 2021-06-24:
Fixed by 8e3abd2 (depends on #4194). I've also uploaded an updated Sculpt 21.03 Falkon package. When trying to play a YouTube video and the video does not start (endless rotating circle, seems to always happen when changing the video resolution), it often helps to reload the page (F5). Some videos (or video players on other web sites) might not work at all yet because of currently disabled proprietary codecs (among other possible reasons).

chelmuth, 2021-06-25:
Related to #3769.

nfeske, 2021-06-25:
>When trying to play a YouTube video and the video does not start (endless rotating circle, seems to always happen when changing the video resolution), it often helps to reload the page (F5).

That is a huge step. I gave the new version a try with youtube and it indeed worked after pressing F5. It is still jittery from time to time, presumably related to the network-performance limitations that were also observed by @jschlatow. But the fact that the video is playing happily is a fantastic achievement already. Congratulations!

commit 464f28e1 (2021-06-28): qt5: implement 'SyncSocket' with pipes on Genode

Fixes #4208
commit a856bfb4 (2021-07-28): qt5: implement 'SyncSocket' with pipes on Genode

Fixes #4208
closed 2021-07-28 by chelmuth