genodelabs/genode #4209: depot_autopilot: bad test timeout

m-stein, 2021-06-25:
3 to 6 times a month we observe a depot autopilot test failing on an arm platform with qemu and hw  (qemu_zynq, pbxa9) because of an unusual short test timeout that is reported afterwards as something like 4300 seconds (near 32bit overflow for usecs). According to the log timestamps the test normally ran much less time than the actual timeout he is configured with.

chelmuth, 2022-09-08:
Happened again last night on *qemu/arm_v7a.pbxa9.hw*

```
[2022-09-08 05:37:32] [init -> depot_autopilot] --- Run "test-timer" (max 30 sec) ---
[2022-09-08 05:37:32] [init -> depot_autopilot] 
[2022-09-08 05:37:32] [init -> depot_autopilot] 0.114 [init -> client] --- timer test ---
[2022-09-08 05:37:32] [init -> depot_autopilot] 0.130 [init -> client] register 4-seconds timeout...
[2022-09-08 05:37:36] [init -> depot_autopilot] 4.133 [init -> client] timeout fired - 0/153/160
[2022-09-08 05:37:38] [init -> depot_autopilot] 5.338 [init -> client] wait 1/10
[2022-09-08 05:37:38] [init -> depot_autopilot] 
[2022-09-08 05:37:38] [init -> depot_autopilot]  test-timer                      failed   4300.890  timeout 30 sec
```

and in July

```
[2022-07-06 08:23:53] [init -> depot_autopilot]  test-timer                      failed   4300.300  timeout 30 sec
```

m-stein, 2022-09-19:
@chelmuth Thanks for the report! I was able to reproduce this in a minimal setup and am now trying to debug it.

m-stein, 2022-09-22:
This should fix the issue:
4a424a6942 base-hw & cortex_a9: consider timer IRQ unreliable
7c9075f98b base-hw: problems with timeout lib in src recipe

I refrained from fixing the BASE_DIR problem in general and concentrated only on the timeout lib for now. Is there a strong desire that I continue this work?

skalk, 2022-09-22:
When looking at commit https://github.com/genodelabs/genode/commit/4a424a6942031cbd645b7c0429637dbfeb806fb2 I would say the _duration calculation was already wrong before, because `cnt` and `last` got interchanged (compare to other implementations, like generic_timer.cc). In the no-overrun use-case, we've to calculate `cnt - last`.
Now, in your new version, you keep the wrong calculations, but interchange both use-cases: overrun and no-overrun. In case that the current counter is greater than the last one stored before, we have no overrun, but here you use the (probably wrong) overrun calculation.
Before following your approach, would you please try to apply the correct calculation on the original approach using your stripped down scenario, and see whether it is then correct? I assume, this might clarify the "wrongly signaled overrun".

skalk, 2022-09-22:
In case, I expressed myself overly complicated, we might discuss the implementation details tomorrow offline?! But I would refrain from merging the timer commit to staging now.

m-stein, 2022-09-22:
@skalk I'm fine with that.

nfeske, 2022-09-23:
Commit https://github.com/genodelabs/genode/commit/93745e399ce8e49da5a4d72b7bcd394e42a8dab8 fixes the build of users of api/base-nova, in particular the vbox5-nova package. The base-nova API includes the statically linked base.lib.a, which depends on timeout.lib.a, which includes base/internal/globals.h. So this header must be captured by the base-nova API archive.

The commit also consistently applies the REP_INC_DIR and `select_from_repositories` pattern to all variants of the timeout build-description files. @m-stein, or did you use REP_DIR in timeout.mk and timeout.mk (https://github.com/genodelabs/genode/commit/970b28a7494652944e4c6b3372963bcaec2ba477) on purpose?

m-stein, 2022-09-23:
@nfeske Thanks for the fix-up! I wasn't aware of the mechanics of the depot and REP_DIR, so, your modifications are sensible.

m-stein, 2022-09-23:
The result of my offline discussion with @skalk:
* His comment about wrong calculations was due to a misunderstanding. The Cortex A9 timer has a decrementing counter while the Generic Timer has an increasing counter. That said, the current calculations are correct.
* Adapting the Cortex A9 timer in a "dirty" way because of a Qemu bug for all hardware users as well gives me and @skalk a bad feeling. However, we don't have a solution that is  both clean _and_ generic. Therefore, we'd suggest a split off of the "dirty" Qemu-specific and the "clean" hardware-specific implementation.

commit 291d4b29 (2022-09-26): base-hw & cortex_a9: consider timer IRQ unreliable

On some Cortex A9 platforms (Qemu 4.2.1 PBXA9), the IRQ status register is not
reliable. Sometimes, it indicates an IRQ too early, i.e., shortly before the
counter wraps. Therefore we have to accomplish wrap detection via counter
comparison only. We check whether the current counter value is higher than the
start counter value of the current timeout.

However, this implies that we have to take care to always read out the counter
before it hits the max timout value again. And, therefore, the max timeout
value has to be far away from the first  [... cut]
closed 2022-10-12 by chelmuth