genodelabs/genode #4210: intel_fb_drv crashes when connecting external monitor

jschlatow, 2021-06-25:
I just noticed that I can reproducibly crash the intel_fb_drv by connecting an external display to my t490s that hasn't been connected before. The external display is connected via display port to a thunderbolt dock. If I start the laptop with the dock connected, I can disconnect/connect the dock later on without any issues. However, if I boot the laptop disconnected from the dock, the intel_fb_drv crashes as soon as I attach the dock showing something similar to the following screen on both displays, internal and external:

[image]

Since this leaves Sculpt in an unrecoverable state, I extracted the following log using report_dump:
```
[leitzentrale -> nitpicker] Warning: Gui (wm -> runtime -> leitzentrale -> inspect) not enough RAM to preserve buffer content during resize
[drivers -> dynamic -> intel_fb_drv] Warning: ww_mutex_lock already locked
[drivers -> dynamic -> intel_fb_drv] ioremap: return sub range phys 0xa1040000 (size 9216000) to virt 0x21040000
[drivers -> dynamic -> intel_fb_drv] [drm_wait_one_vblank] *WARN* vblank wait timed out on crtc 0
[drivers -> dynamic -> intel_fb_drv]  
[drivers -> usb_drv] dev_info: new SuperSpeed USB device number 2 using xhci_hcd
[drivers -> usb_drv] dev_info: USB hub found
[drivers -> usb_drv] dev_info: 4 ports detected
[drivers -> usb_drv] dev_info: new high-speed USB device number 6 using xhci_hcd
[drivers -> usb_drv] dev_info: USB hub found
[drivers -> usb_drv] dev_info: 5 ports detected
[drivers -> usb_drv] dev_info: new SuperSpeed USB device number 3 using xhci_hcd
[drivers -> usb_drv] dev_info: USB hub found
[drivers -> usb_drv] dev_info: 4 ports detected
[drivers -> usb_drv] dev_info: new high-speed USB device number 7 using xhci_hcd
[drivers -> usb_drv] dev_info: USB hub found
[drivers -> usb_drv] dev_info: 4 ports detected
[drivers -> usb_drv] dev_info: new full-speed USB device number 8 using xhci_hcd
[drivers -> usb_drv] dev_info: new high-speed USB device number 9 using xhci_hcd
[drivers -> dynamic -> intel_fb_drv] Found no encoder for the connector HDMI-A-64
[drivers -> dynamic] Warning: intel_fb_drv: RAM upgrade of Capture failed
[drivers -> dynamic] child "intel_fb_drv" requests resources: ram_quota=9000K
[drivers -> usb_hid_drv] dev_info: input: USB HID v1.11 Keyboard [HID 046a:0079] on usb-usbbus-/input0
[drivers -> usb_hid_drv] dev_info: input: USB HID v1.11 Keyboard [HID 046a:0079] on usb-usbbus-/input1
[runtime -> unix_shell -> terminal] Warning: attempt to handle the same signal context twice (nested)
[drivers -> usb_drv] dev_info: new high-speed USB device number 10 using xhci_hcd
[drivers -> usb_drv] dev_info: USB hub found
[drivers -> usb_drv] dev_info: 2 ports detected
[core] [init -> nitpicker] Warning: Gui (runtime -> wm -> wm -> unix_shell -> terminal -> ) not enough RAM to preserve buffer content during resize
[core] [init] child "nitpicker" requests resources: ram_quota=18446743592673218668
[drivers -> usb_drv] dev_info: new low-speed USB device number 12 using xhci_hcd
[drivers -> usb_drv] dev_info: new SuperSpeed USB device number 4 using xhci_hcd
[drivers -> usb_hid_drv] dev_info: input: USB HID v1.10 Mouse [HID 046d:c069] on usb-usbbus-/input0
[drivers -> usb_drv] dev_info: new full-speed USB device number 13 using xhci_hcd
[drivers -> usb_hid_drv] dev_info: input: USB HID v1.11 Device [HID 17ef:3083] on usb-usbbus-/input3
[drivers -> platform_drv] 00000000:00000002.0 power off
[kernel] [ 0] IOMMU:0xffffffff817e4060 FRR:0 FR:0x6 BDF:0:2:0 FI:0x84280000 (0)
[kernel] [ 0] IOMMU:0xffffffff817e4060 FRR:0 FR:0x6 BDF:0:2:0 FI:0x842ab000 (0)
[kernel] [ 0] IOMMU:0xffffffff817e4060 FRR:0 FR:0xc BDF:0:2:0 FI:0x84316000 (0)
[kernel] [ 0] IOMMU:0xffffffff817e4060 FRR:0 FR:0x6 BDF:0:2:0 FI:0x846d5000 (1)
[kernel] [ 0] IOMMU:0xffffffff817e4060 FRR:0 FR:0x6 BDF:0:2:0 FI:0x843ec000 (0)
[kernel] [ 0] IOMMU:0xffffffff817e4060 FRR:0 FR:0x6 BDF:0:2:0 FI:0x84457000 (0)
[kernel] [ 0] IOMMU:0xffffffff817e4060 FRR:0 FR:0x6 BDF:0:2:0 FI:0x844c2000 (0)
[kernel] [ 0] IOMMU:0xffffffff817e4060 FRR:0 FR:0x6 BDF:0:2:0 FI:0x8452d000 (1)
[kernel] [ 0] IOMMU:0xffffffff817e4060 FRR:0 FR:0x6 BDF:0:2:0 FI:0x8452d000 (2)
[kernel] [ 0] IOMMU:0xffffffff817e4060 - disabling fault reporting
[kernel] [ 0] IOMMU:0xffffffff817e4060 FRR:0 FR:0xc BDF:0:2:0 FI:0x8465b000 (0)
[kernel] [ 0] IOMMU:0xffffffff817e4060 FRR:0 FR:0x6 BDF:0:2:0 FI:0x8417e000 (1)
```

The log from a clean run where I had the dock attached during boot looks pretty similar despite the IOMMU messages from the kernel:
```
[drivers -> dynamic -> intel_fb_drv] Warning: ww_mutex_lock already locked
[drivers -> dynamic -> intel_fb_drv] ioremap: return sub range phys 0xa0840000 (size 9216000) to virt 0x20840000
[drivers -> dynamic -> intel_fb_drv] [drm_wait_one_vblank] *WARN* vblank wait timed out on crtc 0
[drivers -> dynamic -> intel_fb_drv]  
[drivers -> usb_drv] dev_info: new SuperSpeed USB device number 5 using xhci_hcd
[drivers -> usb_drv] dev_info: USB hub found
[drivers -> usb_drv] dev_info: 4 ports detected
[drivers -> usb_drv] dev_info: new high-speed USB device number 13 using xhci_hcd
[drivers -> usb_drv] dev_info: USB hub found
[drivers -> usb_drv] dev_info: 5 ports detected
[drivers -> usb_drv] dev_info: new SuperSpeed USB device number 6 using xhci_hcd
[drivers -> usb_drv] dev_info: USB hub found
[drivers -> usb_drv] dev_info: 4 ports detected
[drivers -> usb_drv] dev_info: new high-speed USB device number 14 using xhci_hcd
[drivers -> usb_drv] dev_info: USB hub found
[drivers -> usb_drv] dev_info: 4 ports detected
[drivers -> dynamic -> intel_fb_drv] Found no encoder for the connector HDMI-A-64
[drivers -> usb_drv] dev_info: new full-speed USB device number 15 using xhci_hcd
[leitzentrale -> nitpicker] Warning: Gui (wm -> runtime -> leitzentrale -> inspect) not enough RAM to preserve buffer content during resize
[runtime -> panel_view] Warning: ack for unknown File_system handle 43
[drivers -> usb_drv] dev_info: new high-speed USB device number 16 using xhci_hcd
[runtime -> settings_view] Warning: ack for unknown File_system handle 43
[runtime -> file_browser_view] Warning: ack for unknown File_system handle 43
[runtime -> network_view] Warning: ack for unknown File_system handle 51
[runtime -> network_view] Warning: ack for unknown File_system handle 59
[runtime -> popup_view] Warning: ack for unknown File_system handle 51
[runtime -> popup_view] Warning: ack for unknown File_system handle 43
[runtime -> popup_view] Warning: ack for unknown File_system handle 59
[leitzentrale -> nitpicker] Warning: Gui (wm -> runtime -> leitzentrale -> inspect) not enough RAM to preserve buffer content during resize
[drivers -> usb_hid_drv] dev_info: input: USB HID v1.11 Keyboard [HID 046a:0079] on usb-usbbus-/input0
[drivers -> usb_hid_drv] dev_info: input: USB HID v1.11 Keyboard [HID 046a:0079] on usb-usbbus-/input1
[drivers -> usb_drv] dev_info: new high-speed USB device number 17 using xhci_hcd
[drivers -> usb_drv] dev_info: USB hub found
[drivers -> usb_drv] dev_info: 2 ports detected
[drivers -> usb_drv] dev_info: new low-speed USB device number 19 using xhci_hcd
[core] [init -> nitpicker] Warning: Gui (leitzentrale -> manager -> fader -> ) not enough RAM to preserve buffer content during resize
[drivers -> usb_drv] dev_info: new SuperSpeed USB device number 7 using xhci_hcd
[drivers -> usb_hid_drv] dev_info: input: USB HID v1.10 Mouse [HID 046d:c069] on usb-usbbus-/input0
[drivers -> usb_drv] dev_info: new full-speed USB device number 20 using xhci_hcd
[drivers -> usb_hid_drv] dev_info: input: USB HID v1.11 Device [HID 17ef:3083] on usb-usbbus-/input3
```

alex-ab, 2021-06-25:
> [drivers -> dynamic] Warning: intel_fb_drv: RAM upgrade of Capture failed
> [drivers -> dynamic] child "intel_fb_drv" requests resources: ram_quota=9000K

At this point the driver is out-of-service. The question is why the memory amount is sufficient during boot with attached connector but not when you attach later.

nfeske, 2022-04-13:
This issue is outdated because the intel framebuffer driver has been replaced by a completely new component by now.

closed 2022-04-13 by nfeske