genodelabs/genode #4214: support for breakpoints in gdb on base-linux

tomga, 2021-07-04:
For some time I've been using instructions from [1] for debugging Genode components on `base-linux`. There was however one important thing that was not working that I (and probably not only me) find crucial for effective debugging: ability to set breakpoints in the code to stop execution in concrete places.

When setting a breakpoint in `gdb` it looked like everything is ok but after attempting to continue execution there was a message that setting breakpoint failed. It was disturbing for me enough to force me to look for a solution. And I found the cause and the "fix" for it. The problem comes from setting memory regions for executable code as read only by Genode and effectively stop `gdb` from being able to set a breakpoint in memory. I found that native Linux programs have (at least those that I checked on my system) option MAP_PRIVATE that enables copy on write after modifications of that memory area. Following patch changes mapping options for executable code on `base-linux` and in consequence allow normal debugging sessions for Genode components. 

```
@@ -129,7 +129,8 @@ void *Region_map_mmap::_map_local(Dataspace_capability ds,
 	int  const  fd        = _dataspace_fd(ds);
 	bool const  writable  = _dataspace_writable(ds) && writeable;
 
-	int  const  flags     = MAP_SHARED | (overmap ? MAP_FIXED : 0);
+	int  const  flags     = (executable ? MAP_PRIVATE : MAP_SHARED)
+	                      | (overmap ? MAP_FIXED : 0);
 	int  const  prot      = PROT_READ
 	                      | (writable   ? PROT_WRITE : 0)
 	                      | (executable ? PROT_EXEC  : 0);
```

I've put word "fix" in quotes as I understand that this change is probably not desired to be enabled by default and if enabled with some configuration then this configuration probably must be global for a scenario.

I've tested it on nontrivial scenario without problems but it is possible that in some cases it can cause problems.

As I find it very important for debugging experience I wanted to document it here for reference even if it is not incorporated in Genode although I think there should be some easy way (without changing source directly - some DEFINE?) that would allow to change this behavior.

[1] https://genodians.org/ssumpf/2019-04-26-java-gdb

nfeske, 2021-07-05:
This seems to be a duplicate of #952.

tomga, 2021-07-05:
Indeed. And even more - now I found that @ssumpf mentioned it in its article - I somehow missed that. Sorry for the noise.

closed 2021-07-05 by tomga