genodelabs/genode #4215: problems with SANITIZE_UNDEFINED=yes in libc based components

tomga, 2021-07-04:
Recently I tried to debug some problems in a compoment and tried to use `ubsan` in hope that it will help finding a problem. Unfortunately I did not find a way to use `ubsan` in a component that is `libc` based to not cause crash at early execution.

In 43e38e52c5b97aa2bebc5a020618d12eafb7c9e9 I prepared a sample of my attempt but this code causes crash in sanitizer. It does not understand/like difference `Libc::Env` and `Genode::env` but I don't know what the problem is.

If there are some hard issues with making sanitizer work in such components then I think it should be clearly stated in documentation (I did not see anything like this but I could have missed that). Otherwise I'm asking for some advise how to make it work.

ttcoder, 2021-07-05:
Out of curiosity, was the crash in LD, or libc (did you resolve it to a specific line/method?). I ran into both that as well as failed libc asserts recently, gccaddr2line was useful to help me find what I was doing wrong.

I didn't have trouble with using LibC's flavor of env though, I probably won't be helpful with that (I turn a LibC::Env into a Genode::Env and use it as-is).

chelmuth, 2021-07-05:
I can reproduce this with the following patch and `make -C build/x86_64 run/test PKG=test-ldso KERNEL=linux BOARD=linux`.

```diff
 repos/libports/recipes/pkg/test-ldso/runtime | 6 ++++--
 repos/libports/src/test/ldso/main.cc         | 8 ++++++++
 repos/libports/src/test/ldso/target.mk       | 2 ++
 repos/os/run/test.run                        | 1 +
 4 files changed, 15 insertions(+), 2 deletions(-)

diff --git a/repos/libports/recipes/pkg/test-ldso/runtime b/repos/libports/recipes/pkg/test-ldso/runtime
index 389c7486650..88d83c54d8c 100644
--- a/repos/libports/recipes/pkg/test-ldso/runtime
+++ b/repos/libports/recipes/pkg/test-ldso/runtime
@@ -1,4 +1,4 @@
-<runtime ram="32M" caps="1000" binary="init">
+<runtime ram="128M" caps="1000" binary="init">
 
 	<requires> <timer/> </requires>
 
@@ -110,6 +110,8 @@
 		<rom label="test-ldso_lib_1.lib.so"/>
 		<rom label="test-ldso_lib_2.lib.so"/>
 		<rom label="test-ldso_lib_dl.lib.so"/>
+		<rom label="libsanitizer_common.lib.so"/>
+		<rom label="libubsan.lib.so"/>
 		<rom label="libc.lib.so"/>
 		<rom label="libm.lib.so"/>
 		<rom label="vfs.lib.so"/>
@@ -128,7 +130,7 @@
 		</default-route>
 		<default caps="100"/>
 		<start name="test-ldso">
-			<resource name="RAM" quantum="3M"/>
+			<resource name="RAM" quantum="96M"/>
 			<config ld_bind_now="no" ld_verbose="no">
 				<vfs> <dir name="dev"> <log/> </dir> </vfs>
 				<libc stdout="/dev/log"/>
diff --git a/repos/libports/src/test/ldso/main.cc b/repos/libports/src/test/ldso/main.cc
index 7bd61a44e3c..949c42900fd 100644
--- a/repos/libports/src/test/ldso/main.cc
+++ b/repos/libports/src/test/ldso/main.cc
@@ -187,11 +187,17 @@ static void test_shared_object_api(Env &env, Allocator &alloc)
 }
 
 
+extern void sanitizer_init(Genode::Env &);
+extern void sanitizer_exit();
+extern "C" void wait_for_continue();
+
 /**
  * Main function of LDSO test
  */
 void Libc::Component::construct(Libc::Env &env)
 {
+	sanitizer_init(env);
+
 	static Heap heap(env.ram(), env.rm());
 
 	using Genode::log;
@@ -228,6 +234,7 @@ void Libc::Component::construct(Libc::Env &env)
 
 	log("Catch exceptions in program");
 	log("---------------------------");
+wait_for_continue();
 	try {
 		Rom_connection rom(env, "unknown_rom");
 		error("undelivered exception in remote procedure call");
@@ -277,5 +284,6 @@ void Libc::Component::construct(Libc::Env &env)
 	log("Destruction");
 	log("-----------");
[... 33 lines cut ...]
(gdb) bt
#0  (anonymous namespace)::getVtablePrefix (Vtable=0x0) at .../genode/contrib/sanitizer-58ab991bee9d68f213ae71f04c796de0490f3b0e/src/lib/sanitizer/ubsan/ubsan_type_hash_itanium.cpp:201
#1  __ubsan::checkDynamicType (Object=Object@entry=0x401ff4e0, Type=0x1023060 <typeinfo for Genode::Connection<Genode::Rom_session>>, Hash=15955311242282006732) at .../genode/contrib/sanitizer-58ab991bee9d68f213ae71f04c796de0490f3b0e/src/lib/sanitizer/ubsan/ubsan_type_hash_itanium.cpp:221
#2  0x0000000001901447 in HandleDynamicTypeCacheMiss (Data=0x102ee60 <.Lubsan_data1194>, Pointer=1075836128, Hash=<optimized out>, Opts=...) at .../genode/contrib/sanitizer-58ab991bee9d68f213ae71f04c796de0490f3b0e/src/lib/sanitizer/ubsan/ubsan_handlers_cxx.cpp:36
#3  0x0000000001901adf in __ubsan::__ubsan_handle_dynamic_type_cache_miss (Data=Data@entry=0x102ee60 <.Lubsan_data1194>, Pointer=Pointer@entry=1075836128, Hash=<optimized out>) at .../genode/contrib/sanitizer-58ab991bee9d68f213ae71f04c796de0490f3b0e/src/lib/sanitizer/ubsan/ubsan_handlers_cxx.cpp:87
#4  0x0000000001011a39 in Genode::Connection<Genode::Rom_session>::session (this=this@entry=0x401ff4e0, parent=..., format_args=format_args@entry=0x101ff58 "ram_quota=%ld, cap_quota=%ld, label=\"%s\"") at .../genode/repos/base/include/base/connection.h:173
#5  0x0000000001012d2f in Genode::Rom_connection::Rom_connection (this=0x401ff4e0, env=..., label=<optimized out>) at .../genode/repos/base/include/rom_session/connection.h:41
#6  0x000000000101882f in Libc::Component::construct (env=...) at .../genode/repos/libports/src/test/ldso/main.cc:239
#7  0x0000000010f2eef2 in Component::construct (env=...) at .../genode/repos/libports/src/lib/libc/internal/kernel.h:398
#8  0x00000000500b40b7 in Linker::Binary::call_entry_point (this=0x501add10 <unmanaged_singleton<Linker::Binary, 8, Genode::Env&, Genode::Heap&, Linker::Config const&, char const*>(Genode::Env&, Genode::Heap&, Linker::Config const&, char const*&&)::object_space>, env=...) at .../genode/repos/base/src/lib/ldso/main.cc:442
#9  0x00000000500bad0f in Component::construct (env=...) at .../genode/repos/base/src/lib/ldso/main.cc:843
#10 0x000000005004d5c2 in Genode::Meta::call_member<Genode::Meta::Empty, (anonymous namespace)::Constructor_component, Genode::Meta::Empty> (func=(void ((anonymous namespace)::Constructor_component::*)((anonymous namespace)::Constructor_component * const)) 0x500488a0 <(anonymous namespace)::Constructor_component::construct()>, server=...) at .../genode/repos/base/include/util/meta.h:462
#11 (anonymous namespace)::Constructor::Rpc_construct::serve<(anonymous namespace)::Constructor_component, Genode::Meta::Empty> (args=<synthetic pointer>..., server=...) at .../genode/repos/base/src/lib/base/entrypoint.cc:293
#12 Genode::Rpc_dispatcher<(anonymous namespace)::Constructor, (anonymous namespace)::Constructor_component>::_do_serve<(anonymous namespace)::Constructor::Rpc_construct> (args=<synthetic pointer>..., this=<optimized out>) at .../genode/repos/base/include/base/rpc_server.h:146
[... cut ...]
#18 Genode::Rpc_entrypoint::entry (this=0x500e8da8 <Genode::bootstrap_component()::startup+360>) at .../genode/repos/base/src/lib/base/rpc_dispatch_loop.cc:68
#19 0x000000005008f7d5 in Genode::Thread::_thread_start () at .../genode/repos/base-linux/src/lib/base/thread_linux.cc:80
#20 0x00000000500a5c60 in thread_start () at .../genode/repos/base-linux/src/lib/syscall/spec/x86_64/lx_clone.S:62
```

@cproc could this be a regression after the GCC update? I must admit, I do not know the magic behind ubsan to even guess where the nullptr in `Pointer` on stack is coming from.

tomga, 2021-07-05:
> If there are some hard issues with making sanitizer work in such components then I think it should be clearly stated in documentation (I did not see anything like this but I could have missed that). Otherwise I'm asking for some advise how to make it work.

I just read what I wrote yesterday evening and I see it sounds very harsh. I apologize for it. It was not my intention. It seems I should not write anything so late.

> could this be a regression after the GCC update?

I believe (although I'm not 100% sure) that I've seen it before recent toolchain update.

cproc, 2021-07-06:
The crash got introduced with the 19.05 tool chain update and is related to these upstream changes (which got transferred to GCC):

https://github.com/llvm-mirror/compiler-rt/commit/bfeededd7ee43fe38262619ec200687875ae82a1
https://github.com/llvm-mirror/compiler-rt/commit/6f8de12f02063d5900c639d46dd34788e4c0fd82

So, before the update there was a check for `!Vptr` and now `IsAccessibleMemoryRange()` is called instead, which is currently not implemented for Genode.

Commit 5f60c46 is an attempt to implement this function for Genode by trying to attach a dataspace at the range to be checked and evaluating the result (success or region conflict). This seems to work in principle, but on base-linux an error message is printed each time a region conflict occurs with this check [1] and this error message also causes a recursive call of `Genode::Log::_acquire()` with the ldso test, which leads to another crash (does not happen on base-nova).

Any ideas for a better solution other than removing the base-linux region conflict error message?

[1] https://github.com/genodelabs/genode/blob/ece33d37f834d8bef70458a22f6aee9f3a5d7b8f/repos/base-linux/src/lib/base/region_map_mmap.cc#L151-L155

nfeske, 2023-06-16:
This issue has been stale for two years now.

I wonder, is it still relevant? If nobody plans to pick it up, I propose to close it.

tomga, 2023-06-21:
Hi @nfeske, I think that at that time I tried to use `ubsan` to find some issue with some component. Recently I'm not able to find more time for Genode, than just observing tremendous progress Genode team does, so I won't be able to get back to this issue anytime soon. Given that and fact that there was no interest for a quite long time, I think you can close it.

nfeske, 2023-06-22:
Thanks @tomga for responding and also for the nice compliment. :-)

closed 2023-06-22 by nfeske