genodelabs/genode #4216: keyboard events not routed to window unless a mouse button is pressed

jschlatow, 2021-07-08:
Today, I was able to consistently reproduce an issue occasionally observed by @chelmuth. The observed effect is that Sculpt can be brought into a state in which keyboard events appear to be muted in any window. Yet, @chelmuth observed that, once this state has been reached, key presses/releases are sent to the window when holding the left mouse button.

The described issue occured to me when booting a fresh Sculpt 21.03b image on my x260. While trying to narrow this down, I also managed to consistently reproduce this on both laptops (t490s and x260) using the following instructions:

* make sure you don't have a default GENODE partition
* boot a Sculpt 21.03b image
* activate wired network
* use ram_fs
* select genodelabs depot
* install qt5_textedit and connect it to the system GUI server

Interestingly, if I unuse the ram_fs and use the GENODE partition containing my work setup, the effect vanishes. Even more interestingly, I could switch back to the 'wrong' state by removing every component from my 'default' system, unusing the file system and using the ram_fs again.

nfeske, 2021-07-08:
> install qt5_textedit and connect it to the system GUI server

This seems to be the problem.

In order to direct keyboard events directly to qt5_textedit without a keyboard-focus-handling component, the GUI session of qt5_textedit must be routed to the "keyboard focus" GUI service (which is normally assigned to the window manager).

jschlatow, 2021-07-08:
@nfeske, thanks for the info I was not aware of that. The issue first occured while using themed_wm and I later tried to come up with a minimal test case. I'll double check if I can reproduce the issue with themed_wm as well.

jschlatow, 2021-07-08:
Alright, this indeed seems to be a layer 8 (aka PEBKAC) issue. When first wiring up the themed_wm, I must have clicked the system GUI server instead of the keyboard focus and also made the generated deploy config persistent afterwards. m(

closed 2021-07-08 by jschlatow

nfeske, 2021-07-08:
Layer 8 made my day. :-)