genodelabs/genode #4217: base-hw: introduce main object

m-stein, 2021-07-08:
The base-hw kernel uses several global static variables and "unmanaged singletons", most prominently the data lock, the cpu pool, the first core thread or the kernel stack area. Part of the C++ dialect of the Genode project is avoiding such global data wherever possible. Therefore, I'd opt for introducing a Kernel::Main class and making the mentioned objects data members of it. As a side effect of this issue, we might get rid of the unmanaged-singleton header in the base internal includes.

m-stein, 2021-08-16:
Now this commit series is ready for merging and implements the core of what was requested by this issue:
914fd4ca3e base-hw: get rid of static perf counter object
e03dac0a20 base-hw: serial output as Main member
406e99d61e base-hw: get rid of static variable in Kernel::Pd
3b4ef17c17 base-hw: global IRQ controller as Main member
b8dbef1b7d base-hw: Address-Space-ID allocator as Main member
8377b32bec base-hw: no unmanaged_singleton in kernel thread
1aa5e53c70 base-hw: Core PD as Main member
673e348c8c base-hw: unnecessary unmanaged-singleton includes
7076f87ca0 base-hw: global IRQ pool as Main member
9c1045038e base-hw: don't use boot_info() outside main.cc
dc0687cc14 base-hw: Core main-thread as Main member
05d97a17f8 base-hw: kernel CPU-pool as Main member
34de85f1f4 base-hw: kernel-data lock as Main member
812aa93ed6 base-hw: introduce Kernel::Main class
381c7932ea base-hw: conform pc board header
42e0ac0d07 base-hw: rename Core_thread Core_main_thread
74f742e69b base-hw: get rid of cpu_pool() in platform.cc
62b343049d base-hw: make Trace_source in Platform() a class
09555a24db base-hw: communicate kernel irqs via boot info
3a59663682 base-hw: separate variants of Kernel_object(...)
bd685c17c3 base-hw: always use 'unsigned' for priorities
28db5a21e3 base-hw: fix self-include in kernel/types.h

m-stein, 2021-08-16:
However, there are some minor points that remain:
* There is the arm_v7/virtualization specific ``host_context`` per CPU - global statics that should rather be a member of the CPU objects. However, I would hand in this change later, so, that the above commits can be exposed to nightly tests already.
* The ARM distributors are predestined to be moved to the new ``Global_interrupt_controller`` class (technically not necessary but regarding comprehensibility an improvement)
* As the counter part to the ``Global_interrupt_controller`` the ``Pic`` class could be renamed ``Local_interrupt_controller`` throughout the kernel.
* The kernel applies an approach of buffering values in local static variables in several places ( see ``mp`` in ``src/core/spec/arm_v7/cpu_support.h``, or ``first`` in ``src/include/hw/spec/riscv/page_table.h``, or ``static Page_flags flags`` in ``src/include/hw/spec/x86_64/page_table.h``). I was unsure whether we should stick to this approach but would at least question whether these "buffers" bring any performance improvement.

m-stein, 2021-08-16:
Sorry, I had to slightly adapt the original series (only minor stuff - missing include guards in main.h, a missing include in main.h, the order of data and function members in main.cc):
9644418270 base-hw: get rid of static perf counter object
7db27738c6 base-hw: serial output as Main member
6f83298fa1 base-hw: get rid of static variable in Kernel::Pd
29d81a5d03 base-hw: global IRQ controller as Main member
8b6f5cacb3 base-hw: Address-Space-ID allocator as Main member
9cbe3922e6 base-hw: no unmanaged_singleton in kernel thread
bbdea9b19b base-hw: Core PD as Main member
a336064da3 base-hw: unnecessary unmanaged-singleton includes
818b1e5003 base-hw: global IRQ pool as Main member
20d77a35bd base-hw: don't use boot_info() outside main.cc
06edc743c0 base-hw: Core main-thread as Main member
55da4b3163 base-hw: kernel CPU-pool as Main member
5a8d631837 base-hw: kernel-data lock as Main member
f4936654cb base-hw: introduce Kernel::Main class
381c7932ea base-hw: conform pc board header
42e0ac0d07 base-hw: rename Core_thread Core_main_thread
74f742e69b base-hw: get rid of cpu_pool() in platform.cc
62b343049d base-hw: make Trace_source in Platform() a class
09555a24db base-hw: communicate kernel irqs via boot info
3a59663682 base-hw: separate variants of Kernel_object(...)
bd685c17c3 base-hw: always use 'unsigned' for priorities
28db5a21e3 base-hw: fix self-include in kernel/types.h

m-stein, 2021-08-30:
I just incorporated another series of fixups as the review was stalled anyway and I had some new ideas while rebasing Spunky:
b47c6fd570 base-hw: get rid of static perf counter object
c86296987d base-hw: serial output as Main member
c1b7b3493b base-hw: get rid of static variable in Kernel::Pd
d011468ece base-hw: global IRQ controller as Main member
a5a5441b00 base-hw: Address-Space-ID allocator as Main member
042c1022d0 base-hw: no unmanaged_singleton in kernel thread
b7970c112c base-hw: Core PD as Main member
453744b569 base-hw: unnecessary unmanaged-singleton includes
c4eddd0e2e base-hw: global IRQ pool as Main member
c2ab13901f base-hw: don't use boot_info() outside main.cc
5e8911cfd9 base-hw: Core main-thread as Main member
8aff1a3f02 base-hw: kernel CPU-pool as Main member
076cecad72 base-hw: kernel-data lock as Main member
424ede3f95 base-hw: introduce Kernel::Main class
5c03099ef0 base-hw: conform pc board header
2a4722932f base-hw: rename Core_thread Core_main_thread
870fb5466a base-hw: get rid of cpu_pool() in platform.cc
80fd4f7bb8 base-hw: make Trace_source in Platform() a class
dd2814c5d6 base-hw: communicate kernel irqs via boot info
7500761ca1 base-hw: separate variants of Kernel_object(...)
1d96c27367 base-hw: always use 'unsigned' for priorities
935d215842 base-hw: fix self-include in kernel/types.h

skalk, 2021-08-30:
@mstein I very much appreciate the carefully prepared commit series you've provided. Thank you for dealing with all those grown legacy-code so diligently! I can imagine that it was tiresome at different stages.
The only objection I have is related to commit 042c102 that I have commented inline. Apart from that I'm absolutely fine with merging your commits to staging.

m-stein, 2021-08-31:
@skalk Thanks, I have managed to rework 042c102 using ``alignas`` only. But the change requires two base commits  b811c6992b and 5c2459b5e0 that I think are also reasonable in a more general context.

m-stein, 2021-08-31:
PS: The new thread commit is 3462f8f864.

skalk, 2021-08-31:
@m-stein I'm fine with the new utcb-related commit, thank you. Regarding the two base commits that has to be decided by @nfeske and @chelmuth. If both dependent commits get accepted, we can merge your commit series to staging from my point of view.

chelmuth, 2021-09-02:
I just merged preparatory commits 1865eaff5dbcbb8ac12450975632bd9500b4fe68 and 3798717c50c592f94d84e946a1e52a0ed7a180f2 for builder testing.

nfeske, 2021-09-24:
@m-stein please excuse the delay but I have finally reviewed and merged your base-hw cleanup work.

What to say? Your commit series is feast for the reviewer's eyes! Thank you for putting so much care into splitting the changes into digestible pieces, served in a sensible order, and documented in a way that I can follow. Along the way, you improved many details, like enhancing comments, or improving the consistency of the coding style. That's very much appreciated.

m-stein, 2021-09-24:
@nfeske No problem at all! I'm happy that you found the time now to review and merge. And thank you for your feedback and all the kind words!

nfeske, 2021-09-24:
I just noticed that this change requires the adjustment of each board.h hosted outside the Genode repository such as the one in genode-allwinner. The change comes down to adding an include:

```
#include <spec/arm_v8/address_space_id_allocator.h>
```

However, I'm wondering about the rationale behind defining this type per individual board? Could it alternatively be defined per CPU architecture, relieving the board-support maintainers from adjusting each board.h file?

m-stein, 2021-09-24:
@nfeske As far as I understood it, the plan behind the board headers was to act as single front end for all hardware specific includes, so, that generic code has to include only ``<board.h>``. That's the reason I put it there.

> defining this type per individual board

AFAIS, the allocators are defined arch specific like in your example for ``arm_v8``. However, I see your problem with the inconvenience for developers. Its the question if I got the ``board.h`` concept right and whether we want to prioritize it over this inconvenience.

chelmuth, 2021-09-27:
Affected packages in world.git are
```
make[1]: Target 'genodelabs/bin/arm_v7a/base-hw-odroid_xu/2021-09-26' not remade because of errors.
make[1]: Target 'genodelabs/bin/arm_v7a/base-hw-arndale/2021-09-26' not remade because of errors.
make[1]: Target 'genodelabs/bin/arm_v7a/base-hw-panda/2021-09-26' not remade because of errors.
```

nfeske, 2021-09-28:
Just letting you know that I temporarily removed the commit series from staging.

m-stein, 2021-09-28:
@nfeske Thanks!

m-stein, 2021-09-28:
I'll comment as soon as a new series is ready.

m-stein, 2021-09-28:
Here is a new series for genode.git:
5793e6f214 base-hw: reduce includes in board.h
b0de395805 base-hw: reduce hardware-specific include paths
92f7b38d53 base-hw: don't include board.h from hardw. headers
93c51852f1 base-hw: get rid of static perf counter object
6d090f5d83 base-hw: serial output as Main member
f5211e8844 base-hw: get rid of static variable in Kernel::Pd
1d7d5d394c base-hw: global IRQ controller as Main member
0c8a8a4108 base-hw: Address-Space-ID allocator as Main member
05f54c003f base-hw: no unmanaged_singleton in kernel thread
a4ece71f54 base-hw: Core PD as Main member
cfb7bd831c base-hw: unnecessary unmanaged-singleton includes
3f36b13031 base-hw: global IRQ pool as Main member
5b8f8c273c base-hw: don't use boot_info() outside main.cc
665a8b31fc base-hw: Core main-thread as Main member
85a470fbc4 base-hw: kernel CPU-pool as Main member
690972dfca base-hw: kernel-data lock as Main member
b3f35d35dc base-hw: introduce Kernel::Main class
0cdd2cd86c base-hw: conform pc board header
933ca8b411 base-hw: rename Core_thread Core_main_thread
7624865a61 base-hw: get rid of cpu_pool() in platform.cc
44b90e3485 base-hw: make Trace_source in Platform() a class
3c7d225342 base-hw: communicate kernel irqs via boot info
15c3054c3d base-hw: separate variants of Kernel_object(...)
53f348717d base-hw: always use 'unsigned' for priorities
aaa9f88b93 base-hw: fix self-include in kernel/types.h

And for genode-world.git:
m-stein/genode-world@151239e base-hw: reduce includes in board.h
m-stein/genode-world@33d1325 base-hw: reduce hardware-specific include paths
m-stein/genode-world@c73fe63 base-hw: global IRQ controller as Main member
m-stein/genode-world@1c5bd86 base-hw: Address-Space-ID allocator as Main member
m-stein/genode-world@17ad406 base-hw: kernel CPU-pool as Main member

The new version ensures that all base-hw platforms in world compile as well, has all former fixups applied, including the formerly missing fixups that arose from the review of @skalk, and it prevents that ``address_space_id_allocator.h`` is included in ``board.h``.

nfeske, 2021-09-29:
@skalk Before I merge the new series to staging, I'd like to know do you approve?

m-stein, 2021-09-29:
@skalk All commits except the top three of genode.git (92f7b38, b0de395, 5793e6f) and all of genode-world.git should be the same as during your last review.

m-stein, 2021-09-29:
... not considering a small style fixup of @nfeske (39d25228ea) that has been merged into 0c8a8a4.

skalk, 2021-09-29:
@m-stein thank you very much for the three additional clean-up commits. I've already expected them to be the add-ons after our previous discussion. I very much like it. Although, there are still some corners left, you've already mentioned them in your commit messages, I think the direction is clear and promising. Thank you for taking the time, although it was not on your agenda.

@nfeske I'm fine with the commit series.

nfeske, 2021-09-29:
Thanks @skalk for the feedback. I have merged the new commit series to staging.

m-stein, 2021-09-29:
@skalk Thanks for the review! I also like the direction and see more open issues but had to hold back myself :)

chelmuth, 2021-10-04:
It seems the series breaks the *test-init_loop* and *cpu_quota* tests.

chelmuth, 2021-10-04:
Sorry, I was over-hasty. Despite the error about "unexpected quota distribution", the *test-init_loop* regression seems to be caused by 78e4c423c7c4221ca91b96f8dbea55a42dbd5845 report_rom: versioning and explicit notification.

m-stein, 2021-10-04:
Thanks for the feedback. I'm at debugging the ``cpu_quota`` test.

m-stein, 2021-10-04:
Here's a fixup:
2a384d03d2 fixup "base-hw: always use 'unsigned' for priorities" (fix run/cpu_quota)

closed 2021-10-14 by chelmuth