genodelabs/genode #4220: libc_vfs_fs_fat: uncaught exception

m-stein, 2021-07-13:
Initially, I thought this issue would be more complex because I was on the wrong branch. However, it's only about the calling of a static constructor (staging, x86_64):
``make run/libc_vfs_fs_fat KERNEL=linux|hw|nova BOARD=linux|pc``
```
[init -> vfs] Error: Component::construct() returned without executing pending static constructors (fix by calling Genode::Env::exec_static_constructors())
[init -> vfs] Error: Uncaught exception of type 'Linker::Fatal'
[init -> vfs] Warning: abort called - thread: ep
```
Furthermore, the test is not in the autopilot list which I'd like to change.

m-stein, 2021-07-13:
With these patches...
72c3678067 run/libc_vfs_fs_fat: add to autopilot
16ac9e6172 server/vfs: execute static constructors

... I have tested qemu/x86_64, linux/x86_64, pbxa9. For real hardware, however, the test has built in a fail stop to prevent execution in an environment without a prepared block device. Admittedly, I don't feel competent enough to fiddle around with our nightly hardware in order to bring this test to more targets.

m-stein, 2021-07-13:
Furthermore, shall I try to get rid of the static constructor in the <fatfs> plugin, so, we can prevent 16ac9e6?

m-stein, 2021-07-13:
We could, of course, also use ``vfs_block`` on a disk-image boot module in general to free the test from all the hardware specifics.

m-stein, 2021-07-13:
If I try to add use the ``<fatfs>`` plugin with a minimal POSIX app it gives me the following runtime error:
```
[init -> vfs_example_5] FAT file system "libc_vfs" mounted
[init -> vfs_example_5] Error: Uncaught exception of type 'Genode::My_reconstructible<Fatfs::Platform>::Deref_unconstructed_object'
```
I found that until the FS was successfully mounted, the object ``namespace Fatfs { static Constructible<Platform> _platform; }`` remains constructed and can be used just fine. After that something sets the ``_constructed`` member in ``Constructible`` to false without using any reference to ``_platform`` or the ``Constructible`` interface. The rest of the ``Constructible`` data, however, remains the same.

If I move the plugin to a VFS server and connect from my POSIX app via the ``<fs>`` plugin, everything works fine.

nfeske, 2021-07-14:
I agree with @ssumpf. The actual solution would be removing the global constructor from the VFS plugin. I'd very much appreciate that.

m-stein, 2021-07-14:
With 1c265dd804, libc_vfs_fs_fat succeeds as well as my POSIX application with ``<vfs><fatfs></vfs>``. I must admit that from a technical point, the new solution is not cleaner. However, it prevents the exception and I can't see a better way so far.

I also stumbled over an interesting construct in the fatfs plugin:
```
struct Fatfs_factory : Vfs::File_system_factory {
	struct Inner : Vfs::File_system_factory {
                ...
		Vfs::File_system *create(...) override { 
			return new (env.alloc()) Fatfs::File_system(env, node);
		}
	};
	Vfs::File_system *create(...) override {
		static Inner factory(vfs_env.env(), vfs_env.alloc());
		return factory.create(vfs_env, node);
	}
};
extern "C" Vfs::File_system_factory *vfs_file_system_factory(...) {
	static Fatfs_factory factory;
	return &factory;
}
```
I wonder whether there is a reason for having two nested derivations of Vfs::File_system_factory? With 9e40f05ef4, libc_vfs_fs_fat seems to runs just fine.

m-stein, 2021-07-14:
I found that there is no label attribute to the <fatfs> plugin. The plugin rather seems to use always "0" as label and different occurencies of ``<fatfs>`` all use the same single block session. Is this desirable?

m-stein, 2021-07-14:
Ok, my fault. There is the ``drives`` attribute to control the session label. However, that attribute doesn't have any effect in my setup with a VFS block server as back end. The session label is always "0" because the ``drive`` value is first handed over to the contrib code and then the contrib code provides argument ``label == "0"`` to the function that creates the Block connection.

commit 9f380110 (2021-07-28): run/libc_vfs_fs_fat: add to autopilot

Excludes board 'virt_qemu' from the test as we don't have block device drivers
for this board.

Fixes #4220
commit 74993945 (2021-07-29): lib/fatfs: get rid of global static constructors

There was one global static constructor:

! namespace Fatfs { static Constructible<Platform> _platform; }

This caused applications that used the lib or the <fatfs> VFS plugin to end up
in an uncaught exception due to Genode::Component complaining that method
'construct' returned without executing pending static constructors if they
didn't call Genode::Env::exec_static_constructors().

As the use of Genode::Env::exec_static_constructors() is discouraged in Genode,
this commit rather moves the '_platform' object to the scope of the
initializing  [... cut]
commit 60c83697 (2021-08-18): lib/fatfs: get rid of global static constructors

There was one global static constructor:

! namespace Fatfs { static Constructible<Platform> _platform; }

This caused applications that used the lib or the <fatfs> VFS plugin to end up
in an uncaught exception due to Genode::Component complaining that method
'construct' returned without executing pending static constructors if they
didn't call Genode::Env::exec_static_constructors().

As the use of Genode::Env::exec_static_constructors() is discouraged in Genode,
this commit rather moves the '_platform' object to the scope of the
initializing  [... cut]
closed 2021-08-18 by chelmuth