genodelabs/genode #4224: tool/boot: update bender for balanced intel hwp

m-stein, 2021-07-16:


m-stein, 2021-07-16:
Fixed by 2036f57585.

alex-ab, 2021-07-16:
Please adjust tool/boot/README so that we track the sources matching the binary

m-stein, 2021-07-27:
Added a fixup:
d319e9f5ff fixup "tool/boot: update bender for balanced intel hwp"

chelmuth, 2021-07-28:
> I'm thinking about moving back to "performance" but I'd like to investigate a bit more and, if I switch, I want to make it easily switchable for the user. Therefore this will receive an extra issue.

If you plan to drop the use of this feature, shouldn't we refrain from merging it?

m-stein, 2021-07-29:
I wouldn't drop any feature. For now, the "balanced" setting is fine for me and important for @jschlatow. As soon as I find the time to look deeper into whether the slight performance decrease in my current system really comes from lower CPU frequency levels in moments of high demand, I may want to make the HWP setting configurable to the user, so @jschlatow can set it to "balanced" and I can set it to "performance" with the same bender binary.

chelmuth, 2021-08-24:
At this time I refrain from merging this change as it silently changes the HWP configuration for all users with unpredictable effects in the field. Please consider to make the change configurable with *performance* as default.

I removed the [fixed] tag.

m-stein, 2021-08-24:
Ok, I'll schedule it.

m-stein, 2022-09-05:
@alex-ab Could you give these two commits a review and maybe also test them on a hardware target (if not I'll do the latter tomorrow)?

https://github.com/m-stein/morbo/commit/5086245c62650d4574e19fe961d31fd93b115a98 Make Intel HWP mode configurable via command line 

d3ffb7fce1 Bender: Intel HWP mode configurable via RUN_OPT

alex-ab, 2022-09-06:
Adding the parameter to intel_hwp_16 function will not work. The function is invoked in 16bit mode directly from assembly code (start.asm) and up to now there is no support/arrangement to provide any arguments.

m-stein, 2022-09-06:
Mm, I thought so...

I could test only with qemu up to now and it worked fine. That said, is it right that ``intel_hwp_16`` is not used on every pc platform?

m-stein, 2022-09-06:
Btw., I tried to implement argument support in ``start.asm`` but was not able to find the place where ``intel_hwp_16`` gets called.

m-stein, 2022-09-08:
@alex-ab What do you think of the new version?

alex-ab, 2022-09-08:
@m-stein: I am fine with the bender changes.

commit 9d4a80ea (2022-09-08): Bender: Intel HWP mode configurable via RUN_OPT

The updated Bender version is configurable regarding the mode in which to run
the Intel HWP plugin. This can now be controlled via the new run option
'--bender-intel-hwp-mode' (only when running on NOVA). The option knows 4 valid
values 'off', 'performance', 'balanced', and 'power_saving'. When running on
NOVA, the default value applied by the run tool is 'performance'. when running
on any other kernel, the default value of Bender ('off') is used. Therefore,
when not setting the new '--bender-intel-hwp-mode' flag, the behavior is the
same as bef [... cut]
closed 2022-09-21 by chelmuth