genodelabs/genode #4225: dde_linux: introduce re-newed approach for lx_emul/lx_kit

skalk, 2021-07-19:
Instead of re-implementing the needed datastructures and function declarations, we try to re-use the original Linux kernel headers as far as possible. Moreover, the semantic of the original implementation shall be preserved, e.g. by re-using device driver configuration data from device-tree-blobs. This new approack got tested for various drivers (framebuffer, sd_card, NIC) for the i.MX8 and Allwinner SoC in dedicated repositories. It is time to publish the results at a central place in `repos/dde_linux`.

skalk, 2021-07-22:
@nfeske would you please consider to take fixup commit fd4d7b57fd96537d24dc39e65df598e5c67e5394 as well? There was still a problem in reverting explicitely allocated dataspaces (block request stream buffer) when a session got closed.

skalk, 2021-07-22:
Sorry, I accidentally committed some recipe hash values as well, I've cleaned up the commit. Its id is now 3be10899425933ff0156f95cf6232543f1842e78

nfeske, 2021-07-22:
Sure. I merged https://github.com/genodelabs/genode/commit/3be10899425933ff0156f95cf6232543f1842e78 to staging. Thank you.

commit 1fec1a9a (2021-07-21): dde_linux: introduce new lx_emul/lx_kit library

The re-newed approach currently supports ARM 64-bit only.
It depends on the Platform API of the ARM architecture.
It tries to meet the original semantic of the Linux kernel
functions as far as possible. To achieve this, device drivers
using this library should reference the original Linux kernel
headers at foremost. Only the headers in `src/include/lx_emul/shadow`
have to shadow clone the original ones.

Fix #4225
commit 1a526e73 (2021-07-28): dde_linux: introduce new lx_emul/lx_kit library

The re-newed approach currently supports ARM 64-bit only.
It depends on the Platform API of the ARM architecture.
It tries to meet the original semantic of the Linux kernel
functions as far as possible. To achieve this, device drivers
using this library should reference the original Linux kernel
headers at foremost. Only the headers in `src/include/lx_emul/shadow`
have to shadow clone the original ones.

Fix #4225
closed 2021-07-28 by chelmuth