genodelabs/genode #4227: nic_router: fix NS flag and data offset in Tcp_packet

jschlatow, 2021-07-20:
When trying to log the MSS of TCP packets via `verbose_packets="yes"`, I wondered why the data offset of my TCP packets were always zero and noticed that the NS flag was always set although wireshark claimed the opposite.

Both header fields are found in a single byte of the TCP packet header. Currently, the implementation uses C++ bit field notation (https://github.com/genodelabs/genode/blob/master/repos/os/include/net/tcp.h#L49) to access the fields. Apparently, however, this indicates the compiler to perform a word-wide memory access and, due to the missing endianess conversion, misinterpret the window size as data offset and NS flag.

jschlatow, 2021-07-20:
@m-stein could you please review commit da72d9b?

jschlatow, 2021-07-21:
I also added a commit to correct the name of the CWR flag.

m-stein, 2021-07-21:
Thanks @jschlatow. This looks good to me!

closed 2021-07-28 by chelmuth