genodelabs/genode #4228: vbox5 drops Ethernet packets from usb_modem_drv

jschlatow, 2021-07-20:
When test-running the usb_modem_drv, I noticed that I am unable to open many websites in my browser VM but can surf the same websites with the native falkon.

My investigations showed that vbox5 drops most of the webserver's response packets because they are oversized, i.e. the e1000 device model only supports good-old Ethernet with a maximum of 1522 bytes per frame.
Yet, despite the maximum MTU of 1500, the usb_modem_drv transmits packets of up to 1528 bytes.

As it turned out, the packet size is incorrectly calculated in all dde_linux nic drivers since the Ethernet header size (14 bytes) is added twice (1500 + 2*14 = 1528).

jschlatow, 2021-07-20:
Commit ca70f15 fixes the issue (see commit message).

jschlatow, 2021-07-20:
For future reference, let me also explain why this hasn't occured before and why some websites still worked:

The problem did not occur before because hardware routers typically work with an MTU of 1492 so that the maximum payload of 1500 is never used when connected to a LAN. Since 1492 + 2*14 = 1520, we are still within limits.

Moreover, some webservers (e.g. google, github) use a slightly lower TCP MSS (maximum segment size), which has a similar effect on the maximum observed payload.

chelmuth, 2021-07-21:
That's great news. Thanks for cracking this nut particularly for the detailed description of the effect.

jschlatow, 2021-07-22:
@chelmuth, the original fix messed up the case that the payload is stored in an skb fragment, which was observable with netperf_lwip_wifi.run.
I pushed a fixup commit caafea2 that corrects the size calculation.

chelmuth, 2021-07-22:
Oh my, the lovely Linux skb API. Thanks for the fixup I merged instantly.

closed 2021-07-28 by chelmuth