genodelabs/genode #4230: libusb allocates wrong packet size for control messages

jschlatow, 2021-07-21:
When investigating why the usb_webcam driver wasn't working for my t490s, I found the culprit in the libusb backend.

The t490s has an integrated webcam from "Chicony Electronics" with vid/pid 04f2:b67c. With this webcam, I became USB stall errors when the webcam driver sent a video probe control message. The reason for this was an incorrect wLength field of the control messages. Although the UVC spec defines how to respond to a wLength that is smaller/larger than expected, this particular webcam just did not provide a response at all.

The error lies within the libusb backend, which allocates a too large packet from the packet stream.

chelmuth, 2021-07-22:
Can you provide any information (or pointer to the spec) about the difference between `setup->wLength` (new) and `transfer->length` (old)?

jschlatow, 2021-07-22:
Sure, this is a matter of how libusb handles control messages. In libusb's `libusb_fill_control_transfer())`, the `transfer->length` is increased by `LIBUSB_CONTROL_SETUP_SIZE` because the `transfer->buffer` includes a setup header. This is actually already respected by the memcpy in [L572](https://github.com/genodelabs/genode/blob/2cee7642b0cf5b77ec7b8e2c4cf86c0d9a476e84/repos/libports/src/lib/libusb/genode_usb_raw.cc#L572), which only copies the payload after the setup header. Since the info provided in the setup header is already part of the Packet_descriptor, we must only allocate for the actual payload (i.e. `setup->wLength`).

chelmuth, 2021-07-22:
Thanks, now it's crystal clear.

closed 2021-07-28 by chelmuth