genodelabs/genode #4231: libuvc: add support for UVC 1.50

jschlatow, 2021-07-21:


jschlatow, 2021-07-21:
Commit 6cc9ba9 adds basic UVC 1.50 support to libuvc. Together with the fix from #4230, I get a webcam video stream on my t490s. However, the picture is grayscale, probably due to some missing configuration.

jschlatow, 2021-07-22:
Apparently, the picture only appeared to be grayscale because of a wrong exposure setting that produced a very dark picture. Commit ed2de1f fixes this by correctly enabling one of the auto exposure settings if available.

ssumpf, 2021-07-22:
> Commit [6cc9ba9](https://github.com/genodelabs/genode/commit/6cc9ba9c4aeae6ea50ccce8db191b53e65e667c2) adds basic UVC 1.50 support to libuvc. Together with the fix from #4230, I get a webcam video stream on my t490s. However, the picture is grayscale, probably due to some missing configuration.

@jschlatow: Would  you like to try to get this patch upstream (https://github.com/libuvc/libuvc.git)?

chelmuth, 2021-07-23:
Upstream had an according issue https://github.com/libuvc/libuvc/issues/42 which was closed. Maybe accidentally?

jschlatow, 2021-07-23:
It appears the issue was closed because they were missing a webcam that didn't work with the old spec.
I added a new PR libuvc/libuvc#187.

chelmuth, 2021-07-28:
Merged to master.

closed 2021-07-28 by chelmuth