genodelabs/genode #4234: sculpt_manager: support other storage devices than ahci, nvme as well

skalk, 2021-07-22:
Currently, the sculpt manager decides dependent on the label where to route block sessions to, e.g. when starting `gpt_write` to expand a partition. When it is 'usb' it is routed to the runtime, when it is 'ahci' or 'nvme' it goes to the driver subsystem. I know in the long run this differentiation will be superflous when all drivers are moved to the runtime. For now, I need a solution to support the sd_card driver in imx8 as well. My current solution is to check for the usb label and route everything else to the drivers subsystem. See the upcoming patch.
@nfeske please tell me whether you are okay with this, or what kind of solution you would prefer. Thank you!

nfeske, 2021-07-22:
I think your intermediate solution is fine. I merged your commit to staging.

nfeske, 2021-08-31:
Fixed in master.