genodelabs/genode #4235: nic_router: do not re-use ARP request as reply

m-stein, 2021-07-22:
So far, in order to create an ARP reply, the NIC router merely created a copy
    of the corresponding ARP request and modified only those values that differ.
    This approach has the disadvantage of re-using bad parameters from a broken
    request. The specific use-case that made this visible was an early version of
    the Pine board network driver that used to forward ARP requests with a greater
    size than required. The ARP replies of the router re-used this size and
    confused other network nodes with that. In general, the NIC router should
    rely on the data of incoming packets the least possible.

m-stein, 2021-07-22:
Thanks @nfeske for having reported this and provided a quickfix solution!

Fixed:
fb55ae774b nic_router: do not re-use ARP request as reply

chelmuth, 2021-07-27:
da34733fca15f036157a889deb8794ab92aa257a seems to break *run/nic_dump* and perhaps several other network tests. @m-stein please investigate.

m-stein, 2021-07-27:
@chelmuth Thanks for the hint! I'm sorry, I copied from ARP request code and forgot to adapt the opcode. For some reason, the ``run/nic_router`` script that I used for testing succeeded anyway.

Fixup is 7781743703065ce3d84177725b8208515fc3650f.

commit da34733f (2021-07-26): nic_router: do not re-use ARP request as reply

So far, in order to create an ARP reply, the NIC router merely created a copy
of the corresponding ARP request and modified only those values that differ.
This approach has the disadvantage of re-using bad parameters from a broken
request. The specific use-case that made this visible was an early version of
the Pine board network driver that used to forward ARP requests with a greater
size than required. The ARP replies of the router re-used this size and
confused other network nodes with that. In general, the NIC router should
rely on the data o [... cut]
closed 2021-07-28 by chelmuth