genodelabs/genode #4238: nic_router: replace term "uplink" with "NIC client"

m-stein, 2021-07-27:
The term "uplink" for network interfaces in the router that have a NIC session
    client as back end was introduced in a time when Uplink sessions didn't yet
    exist. Now, they do and, although both an uplink and an Uplink session
    normally describe a network session between router and network device driver,
    they are based on two different service types (NIC and Uplink). This can easily
    cause confusion when integrating the router (the <uplink> is not related to
    Uplink sessions) or trying to understand its functioning (an 'Uplink' object
    has nothing to do with the Uplink service).

m-stein, 2021-07-27:
Fixed:
70da11926c nic_router: replace term "uplink" with "NIC client"

chelmuth, 2021-07-27:
Doesn't this change require adaptions in other scenarios (namely Sculpt) too?

m-stein, 2021-07-27:
@chelmuth No, because drivers are all Uplink clients now (managed via ``<policy>``). Therefore, Sculpt doesn't have to use NIC clients. In general NIC clients are only used whenever the NIC router is stacked behind another NIC router or the the NIC bridge for instance.

m-stein, 2021-07-27:
Commit https://github.com/m-stein/genode-world/commit/2c2953bcdbcf0df354e74a356e43e017d80ec74e is also related.

chelmuth, 2021-07-28:
> @chelmuth No, because drivers are all Uplink clients now (managed via `<policy>`). Therefore, Sculpt doesn't have to use NIC clients. In general NIC clients are only used whenever the NIC router is stacked behind another NIC router or the the NIC bridge for instance.

Thanks for the explanation. Seems I was confused by the many flavors of uplink.

chelmuth, 2021-07-28:
> Commit [m-stein/genode-world@2c2953b](https://github.com/m-stein/genode-world/commit/2c2953bcdbcf0df354e74a356e43e017d80ec74e) is also related.

I expect *src/app/mbimcli/construct.cc* must be adapted as it generates the NIC-router config in line 950.

jschlatow, 2021-07-28:
> > Commit [m-stein/genode-world@2c2953b](https://github.com/m-stein/genode-world/commit/2c2953bcdbcf0df354e74a356e43e017d80ec74e) is also related.
> 
> I expect _src/app/mbimcli/construct.cc_ must be adapted as it generates the NIC-router config in line 950.

That's correct. I also expect some README files to require an update.

This might be a good opportunity to give my new [github search utility](https://github.com/jschlatow/github-search) a try ;-) It's still quite minimalistic, yet already provides an overview where to look closer.

chelmuth, 2021-07-29:
@m-stein I merged your merge_to_staging branch into genode-world/staging as genodelabs/genode-world@6983b307a9e504b6e0be20e871f99cca169fec10

m-stein, 2021-07-29:
@jschlatow I just gave your ghs tool a try and I'm a bit stuck. Here's what I did (Ubuntu 18.04.5 LTS, inside my ghs clone):
```
$ apt install python3
...
$ apt install python3-pip
...
$ python3 -m pip install rich
...
$ python3 -m pip install ghapi
...
$ python3 -m pip install requests
WARNING: Value for scheme.platlib does not match. Please report this to <https://github.com/pypa/pip/issues/10151>
...
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (2.18.4)
WARNING: Value for scheme.headers does not match. Please report this to <https://github.com/pypa/pip/issues/10151>
...
$ ./ghs --config examples/genode.yaml vfs
Traceback (most recent call last):
  File "./ghs", line 8, in <module>
    import yaml
ImportError: No module named yaml

$ python3 -m pip install yaml
WARNING: Value for scheme.platlib does not match. Please report this to <https://github.com/pypa/pip/issues/10151>
...
Defaulting to user installation because normal site-packages is not writeable
ERROR: Could not find a version that satisfies the requirement yaml (from versions: none)
ERROR: No matching distribution found for yaml

$ sudo apt install python3-yaml
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-yaml is already the newest version (3.12-1build2).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
```
Do you know whether I have missed something?

m-stein, 2021-07-29:
@chelmuth Thanks.

I tried to check all occurencies of uplink/Uplink also in README files in Genode and Genode World although I couldn't get ghs to work so far.

jschlatow, 2021-07-29:
@m-stein, thanks for the report. I assume the problem is that Ubuntu still uses Python 2.7 by default. I changed the shebang to  look for a `python3` binary instead of `python`.

m-stein, 2021-07-29:
@jschlatow Thanks. Now it works :)

closed 2021-08-18 by chelmuth