genodelabs/genode #4242: nic router: dns server list in one dhcp option

m-stein, 2021-08-04:
If there are multiple DNS server addresses, the NIC router DHCP server currently adds also multiple DHCP option 6 entries to responses where each option contains one address. This conflicts with https://datatracker.ietf.org/doc/html/rfc2132#section-3.8 which states that the addresses should be listed within one option 6 field without delimiter.

m-stein, 2021-08-05:
Fixed:
ad91f7d9e3 nic_router: make Ipv4_config a class
ec33ed48c0 nic_router: list dns servers in one dhcp option 6

m-stein, 2021-08-05:
I just asked myself whether it would be desirable to let the router's DHCP client still assume multiple option 6 entries (each potentially containing multiple addresses) in DHCP replies. With the above commit, the router would again only parse the first option 6 entry. The Linux IP stack supports parsing multiple entries for some reason that I don't know.

m-stein, 2021-08-09:
Added a fixup:
a86329499e fixup "nic_router: list dns servers in one dhcp option 6"

m-stein, 2021-08-10:
Thanks for that feedback. I'd vote for not adding the functionality until we encounter misbehaving equipment because I'd like to document why it was added.

chelmuth, 2021-08-10:
The RFCs support my assumption that options should occur only once (with exceptions).

> RFC2131:  Options may appear only once, unless otherwise specified in the options document.  The client concatenates the values of multiple instances of the same option into a single parameter list for configuration.

> RFC2132: The domain name server option specifies a list of Domain Name System name servers available to the client. Servers SHOULD be listed in order of preference. [no exeception specified]

m-stein, 2021-08-10:
Thanks for looking this up!

commit f3a78399 (2021-08-10): nic_router: make Ipv4_config a class

The fact that the IPv4 config was a struct with all data members public was a
mere leftover of an early state of the NIC router. Today, the router
implementation style is to avoid structs and public data members wherever
possible.

This commit slightly changes the behavior of the router regarding log output.
The router used to print malformed IPv4 configurations to the log only if
the 'verbose' config flag was set using this style:

! [my_domain] malformed dynamic IP config: interface 10.0.2.1/24 ...

Now, malformed IPv4 configurations are only printed if  [... cut]
closed 2021-08-18 by chelmuth