genodelabs/genode #4243: Wondering about Trace::timestamp() on x86_{32,64}

jschlatow, 2021-08-06:
This week, I came across a White Paper from Intel titled "How to Benchmark Code Execution Times on Intel® IA-32 and IA-64 Instruction Set" that describes different variants of serialising instructions when reading the TSC.
Out of curiosity, I had a look into the `Trace::timestamp()` implementations, which raised a few questions on my side.

1. Both implementations, [x86_32](https://github.com/genodelabs/genode/blob/master/repos/base/include/spec/x86_32/trace/timestamp.h) and [x86_64](https://github.com/genodelabs/genode/blob/master/repos/base/include/spec/x86_64/trace/timestamp.h), use two blocks of inline asm. I am wondering whether it is guaranteed that the compiler doesn't move the second block before the first block. Moving the block would foil the intended serialisation though.
2. I am also wondering why the eax register need to be zeroed before executing cpuid. Since, cpuid is only used for serialisation, the value of eax shouldn't matter. Does it?
3. In the x86_32 implementation, the cpuid operation is framed with a pushl and popl on ebx. Why is that? Can't ebx just be added to the clobber list?

@cnuke, @alex-ab Any chance you have some explanations?

cnuke, 2021-08-06:
Good questions, unfortunately I cannot remember what prompted me to implement it in that way - maybe it is something along the lines of other mechanisms, like `lfence`, being more costly on older generations and/or CPUs from other vendors.

(That being said, I am all for replacing the `cpuid` dance with `lfence` or something similar as it also might prevent undesirable side-effects when running Genode in a VM.)

To answer your questions (probably in an unsatisfactory way):

 1. I do not know if it is guaranteed in any way but I cannot remember GCC reodering the blocks. So that could change with other compilers or flags.
  2. `cpuid` behaves differently depending on the value auf `EAX`. Making sure it is the same every time reduces its variance. How big the impact for this use-case is I am not sure.
  3.

jschlatow, 2021-08-09:
@cnuke, thanks for the quick reply. From the aforementioned (but maybe outdated) whitepaper from Intel, the `cpuid` dance appears to be the recommended move.

Good point with trying to reduce the variance of the `cpuid` instruction by providing a constant operand.
Do you have any argument against merging the asm blocks into a single block just to safely exclude reordering by the compiler?

Re. 3: I was wondering if you left it blank in your reply intentionally?

cnuke, 2021-08-09:
> @cnuke, thanks for the quick reply. From the aforementioned (but maybe outdated) whitepaper from Intel, the `cpuid` dance appears to be the recommended move.

At some pointed I wanted to revisit this topic, especially because others, e.g., Linux, seem to work fine with some `lfence` approach and using `cpuid` in a VM could be costly (as the instruction normally traps). If you would like to look into that be my guest.

> Do you have any argument against merging the asm blocks into a single block just to safely exclude reordering by the compiler?

No, go for it :+1:.

> Re. 3: I was wondering if you left it blank in your reply intentionally?

No, that was quite *unintentional*:

3. Clobbering should work as well - I cannot recall a reason and my notes are silent about that (I prototyped the tracing stuff on Linux, it was probably more obvious at the time).

chelmuth, 2021-08-09:
>  Clobbering should work as well - I cannot recall a reason and my notes are silent about that (I prototyped the tracing stuff on Linux, it was probably more obvious at the time).

IIRC @ssumpf always stated that EBX cannot be clobered in PIC mode as it's used for the GOT pointer.

chelmuth, 2021-08-09:
> Both implementations, [x86_32](https://github.com/genodelabs/genode/blob/master/repos/base/include/spec/x86_32/trace/timestamp.h) and [...]

Does this mean it's not guaranteed that two `asm volatile` blocks aren't reordered without artificial dependencies? Sorry, but I tried to find the answer [here](https://gcc.gnu.org/onlinedocs/gcc-11.2.0/gcc/Extended-Asm.html#index-asm-volatile) but had no luck.

m-stein, 2021-08-09:
[This](https://gcc.gnu.org/onlinedocs/gcc-11.2.0/gcc/Extended-Asm.html#index-asm-volatile) states that...
*... the compiler can move even volatile asm instructions relative to other code, including across jump instructions. For example, on many targets there is a system register that controls the rounding mode of floating-point operations. Setting it with a volatile asm statement, as in the following PowerPC example, does not work reliably.*
(without artificial dependency)

cnuke, 2021-08-09:
> > Clobbering should work as well - I cannot recall a reason and my notes are silent about that (I prototyped the tracing stuff on Linux, it was probably more obvious at the time).
> 
> IIRC @ssumpf always stated that EBX cannot be clobered in PIC mode as it's used for the GOT pointer.

Ah right, that rings a bell even ever so silently.

chelmuth, 2021-08-09:
> [This](https://gcc.gnu.org/onlinedocs/gcc-11.2.0/gcc/Extended-Asm.html#index-asm-volatile) states that...
> _... the compiler can move even volatile asm instructions relative to other code, including [...]

But this still does not clarify if `asm volatile (...); asm volatile (...)` may be reordered. Up to now I naively assumed the compiler is not allowed to do this, but now I'm afraid this assumption is void.

alex-ab, 2021-08-09:
> But this still does not clarify if `asm volatile (...); asm volatile (...)` may be reordered. Up to now I naively assumed the compiler is not allowed to do this, but now I'm afraid this assumption [...]

I think the clobber argument "memory" is important in that context to use. According to the doc the compiler can't assume anything memory related about before nor behind the asm if set:

```
"memory"
The "memory" clobber tells the compiler that the assembly code performs memory reads or writes to items other than those listed in the input and output operands (for example, accessing the memory pointed to by one of the input parameters). To ensure memory contains correct values, GCC may need to flush specific register values to memory before executing the asm. Further, the compiler does not assume that any values read from memory before an asm remain unchanged after that asm; it reloads them as needed. Using the "memory" clobber effectively forms a read/write memory barrier for the compiler.

Note that this clobber does not prevent the processor from doing speculative reads past the asm statement. To prevent that, you need processor-specific fence instructions.
```

jschlatow, 2021-08-13:
I added a few comments and the memory clobber in commit a205459.

closed 2021-08-31 by chelmuth