genodelabs/genode #4244: Trace_buffer wrapper reads last processed entry twice

jschlatow, 2021-08-06:
The `vfs_trace` plugin and the `trace_logger` component use a convenience wrapper (called `Trace_buffer`) around `Trace::Buffer` for iterating new buffer entries.

The `Trace_buffer` class stores the last processed entry in its member `_curr` in order to continue the next iteration at the correct position in the trace buffer. Yet, when starting a new iteration, the functor is called on `_curr` a second time.

jschlatow, 2021-08-06:
@m-stein, I added a commit addressing this issue to my vfs_ctf branch, could you have a look?

m-stein, 2021-08-09:
@jschlatow Sure.

m-stein, 2021-08-10:
Unfortunately, I'm again confused by the design and documentation of ``Genode::Trace::Buffer``. I think that 425df580634b3c8e4dcb85b58c45181e36932843 still contains a bug. When we enter ``for_each_new_entry`` and ``_curr.last()`` yields "true" this might indicate that, last time, we left the function after having processed the (back then) most recently committed entry of the buffer. Let's say this entry was located at buffer address ``x``. Therefore ``((Entry*)((addr_t)x + ((_Entry*)x)->len)).len`` would have yielded 0 back then. However, in the meantime, the buffer might have advanced and the entry next to ``*(Entry*)x`` is now valid, but the current call to ``for_each_new_entry`` would nonetheless jump back to ``buffer.first()`` because of ``_curr`` still being invalid.

m-stein, 2021-08-10:
I'd vote for a rework of ``Trace::Buffer`` because I think it could make the code built based on it way more intuitive. However, I hesitate to do it without having coordinated the approach with the author of ``Trace::Buffer``.

jschlatow, 2021-08-10:
> Unfortunately, I'm again confused by the design and documentation of `Genode::Trace::Buffer`. I think that [425df58](https://github.com/genodelabs/genode/commit/425df580634b3c8e4dcb85b58c45181e36932843) still contains a bug. When we enter `for_each_new_entry` and `_curr.last()` yields "true" this might indicate that, last time, we left the function after having processed the (back then) most recently committed entry of the buffer. Let's say this entry was located at buffer address `x`. Therefore `((Entry*)((addr_t)x + ((_Entry*)x)->len)).len` would have yielded 0 back then. However, in the meantime, the buffer might have advanced and the entry next to `*(Entry*)x` is now valid, but the current call to `for_each_new_entry` would nonetheless jump back to `buffer.first()` because of `_curr` still being invalid.

Note, that `_curr.last()` never yields true unless we have never processed any entry because `_curr` is only updated with valid buffer entries.

m-stein, 2021-08-10:
Sorry, the above suspected "bug" was based on a misconception of mine.

m-stein, 2021-08-10:
@jschlatow Thanks :)

m-stein, 2021-08-10:
It seems the implementation still contains the well known problem that ``_curr`` might refer to an entry that was already overridden with a new one because the buffer wrapped in the meantime. The answer to that so far was that users should choose a buffer large enough to prevent that this can happen.

jschlatow, 2021-08-10:
That's correct. At the moment, however, I don't see how to prevent this. Even with a different Trace::Buffer implementation, we'd need to decide what to do for a full buffer. I believe the current approach of keeping the most recent entries (i.e. overwriting the oldest entry) is quite useful. Sure, the problem remains that we don't know where to find the oldest entry once the buffer wrapped multiple times with the current implementation.

m-stein, 2021-08-10:
Just a humble suggestion from my side:
525c594388 fixup "Trace_buffer: do not reprocess last read entry" (var names)
3127b11e7f fixup "Trace_buffer: do not reprocess last read entry" (doc)

Apart from that I have nothing to add.

jschlatow, 2021-08-10:
@m-stein Thanks for your review and your reasonable additions. I mirrorred your fixups to the trace_logger as well and squashed everything into commit 85767e1.

m-stein, 2021-08-10:
I missed that the file is still mirrored. Maybe we should take the opportunity to get rid of the mirror by providing a central version at, e.g., ``os/include/trace_logger/trace_buffer.h``

jschlatow, 2021-08-16:
What about moving the implementation to `os/include/trace/trace_buffer.h` and creating an api archive for that? The directory, however, already contains the `policy.h` used by the trace policies. I'm not sure whether it is wise to combine the policy api and the monitor api in a single archive.

nfeske, 2021-08-17:
> I'm not sure whether it is wise to combine the policy api and the monitor api in a single archive.

I think this would be fine.

jschlatow, 2021-08-23:
Commit 6100016 contains the consolidation of the trace_buffer.h implementations of trace_logger and vfs_trace. I also added the trace api archive via commit b60ee31.

@chelmuth, I moved all relevant commits to an `issue4244_*` branch which is ready to be merged to staging now.

nfeske, 2021-08-31:
Fixed in master.

nfeske, 2022-02-03:
I'm afraid that commit 1416b2258f8140e9d68307bffd33898f20c94495 introduced a new problem in exchange for one problem solved.

The symptom can be observed with the trace_logger test by adjusting the `period_sec` in repos/os/recipes/pkg/test-trace_logger/runtime to one second (well, this is not really needed but it makes the problem appear a bit quicker), and running the test via:
```
x86_32$ make run/depot_autopilot TEST_PKGS=test-trace_logger KERNEL=okl4 BOARD=pc
```

At each period, one trace-buffer entry is missing. Normally, there is the pattern like this: two RPC names `trigger_once` followed by a counting value.
```
[init -> depot_autopilot] 2.709 [init -> trace_logger] 14 
[init -> depot_autopilot] 2.714 [init -> trace_logger] trigger_once
[init -> depot_autopilot] 2.719 [init -> trace_logger] trigger_once
[init -> depot_autopilot] 2.724 [init -> trace_logger] 15 
[init -> depot_autopilot] 2.729 [init -> trace_logger] trigger_once
[init -> depot_autopilot] 2.734 [init -> trace_logger] trigger_once
[init -> depot_autopilot] 2.739 [init -> trace_logger] 16 
[init -> depot_autopilot] 2.746 [init -> trace_logger] trigger_once
[init -> depot_autopilot] 2.751 [init -> trace_logger] trigger_once
```
However, the subsequent count (in this case 17) is not captured. The start of the next period looks like this:
```
[init -> depot_autopilot] 4.289 [init -> trace_logger] trigger_once
[init -> depot_autopilot] 4.295 [init -> trace_logger] trigger_once
[init -> depot_autopilot] 4.300 [init -> trace_logger] 18 
[init -> depot_autopilot] 4.305 [init -> trace_logger] trigger_once
[init -> depot_autopilot] 4.315 [init -> trace_logger] trigger_once
[init -> depot_autopilot] 4.321 [init -> trace_logger] 19
```

I tracked the problem to this line https://github.com/genodelabs/genode/blob/master/repos/os/include/trace/trace_buffer.h#L74 in `Trace_buffer` utility.

When I remove the `else` branch, the problem disappears. However, given my poor understanding of the code, I'm afraid that this is not a proper fix. @jschlatow could you try reproducing the issue and have a look?

I wonder, might there be a way to express the code of `for_each_new_entry` in a less complex way? To me, the apparent need for the large comment block (which I admittedly struggled following) smells like an opportunity for simplification. ;-)

jschlatow, 2022-02-03:
Sure. It appears the last round of every `for_each_new_entry` calls the functor with an empty entry but still considers the entry processed. The quickfix would be to add a `if (!entry.length()) break`.

I also see huge potential for optimisation. Every time I look into `trace/buffer.h` and `trace_buffer.h` I'm irritated by the meaning of `Trace::Buffer::Entry::last()`. It is somewhere along the lines of an invalid entry but not quite as there are non-last entries with zero-length. I will try to come up with a better solution here by adding a `Trace::Buffer::Entry::valid()`. I also think that `Trace::Buffer::next()` needs some rethinking.

nfeske, 2022-02-03:
Thanks @jschlatow for the quick response. The quick fix indeed solves the problem. I tripped over the meaning of the `_curr` member and think it should be reconsidered to always point at the entry not yet processed, not the last piece of work already done. This would correspond to the pattern used in other contexts like the a file-read loop (tracking the bytes processed = offset to the work to be done in the next round).

Anyway, the quickfix is definitely an improvement worth merging in the meanwhile. I merged it as 32834d8edc2befa60dffcc603a7b1e94f1765a22.

Regarding your last paragraph, yes, we should definitely brainstorm possible improvements. Looking forward to it. :-)

jschlatow, 2022-02-03:
Thinking about this, there are actually a few "problems" with the current buffer implementation. One particular aspect I quite dislike is that there is currently no way to prevent a fast writer from overwriting the entry the reader is processing at the moment. I believe a double-buffering approach could help here to guarantee that the writer has always half a buffer of space to write its entries. However, I need to spend some time on this to think this through.

jschlatow, 2022-02-07:
@nfeske, commit 0bf9c0d fixes the issue that events are not read after the buffer wrapped.

closed 2022-02-15 by nfeske