genodelabs/genode #4245: util/register: registers on unaligned integers

m-stein, 2021-08-09:
The register framework so far doesn't support writing bitfields in registers
    on unaligned integers.

m-stein, 2021-08-09:
Fixed:
0411da928e util/register: registers on unaligned integers

m-stein, 2021-08-09:
@jschlatow This might be interesting for you.

jschlatow, 2021-08-09:
@m-stein, I noticed that I don't even need to use the newly introduced `set_unaligned()`. Just changing the type to `unaligned_access_t` seems to work just as good.

m-stein, 2021-08-10:
Hm, I must admit that I don't understand why ``set_unaligned`` isn't necessary. Should we remove the unaligned methods again? Any opinions?

chelmuth, 2021-08-10:
I must admit I can't follow the fix by @jschlatow too. Does this mean you don't need any patch to the register framework to fix your issue?

jschlatow, 2021-08-10:
Yes, apparently, adding the `aligned(1)` is sufficient for getting rid of the "cannot bind packed field" error from GCC. Yet, I don't have a full explanation either.

chelmuth, 2021-08-16:
Any opinions if we should drop the patch in this issue as the user can just declare the client-side data type as `aligned(1)`?

m-stein, 2021-08-16:
Maybe it's not a good idea to write code based on the fact that the current GCC version seems to deal fine with using the same signature and implementation of a method although the arguments may have different alignment. I would vote for making this difference specific by calling different methods, i.e., merge the new framework feature.

chelmuth, 2021-08-18:
In contrast, I think we should not merge code that addresses potential future issues.

m-stein, 2021-08-18:
@chelmuth AFAICT, the use case of @jschlatow is a present one and fixing it by using an implicit mechanism of the GCC of which we don't have an idea why it works is not what I would opt for.

m-stein, 2021-08-18:
Sorry, this came out a bit more harsh than I meant it. I just wanted to throw in that the issue is a present one.

jschlatow, 2021-08-19:
On a second thought, actually, my use case does not depend on having an API for unaligned register access. My problem originated from the fact that the `packed` attribute implicitly tells the compiler it cannot make any assumptions on alignment. Apparently, GCC takes a conservative approach and prints an error if you want to pass a reference of a packed field since this might lead to illegal instructions on some architectures. Adding the `aligned(1)` attribute, however, specifies the alignment assumptions so that GCC, apparently, can decide how to access the referenced memory appropriately. Since the method is inlined anyway, I would assume that the GCC will make this decision on a per-call basis. This would explain why there is no difference between `set()` and `set_unaligned()`.

In my use case, however, I actually haven't intended to cause any unaligned memory access. In fact, the struct's fields I want to access with the register framework are actually aligned appropriately. I can let the compiler know about this by adding the packed attribute only to the other struct members.