genodelabs/genode #4246: nic_router: propagate dns domain name via dhcp

m-stein, 2021-08-09:
The NIC router DHCP client should read-out the first DNS domain name (DHCP option 15) if any from a DHCP reply that generates an IPv4 config for a domain and store the name together with the IPv4 config for that domain. Furthermore, the NIC router DHCP server should be able to obtain a DNS domain name from another domain with a DHCP client (dynamically) or from its configuration (statically) and propagate this name with DHCP replies (DHCP option 15).

In the course of solving this issue, I would rename the ``dns_server_from`` config attribute of the router to ``dns_config_from`` and change it's semantic in a way that it includes also the domain name of the remote domain. Furthermore, the ``<dhcp-server>`` tag would receive a new sub-tag ``<dns-domain name="..."/>`` for statically defining a domain name.

m-stein, 2021-08-11:
Fixed:
9d50c23065 nic_router: propagate dns domain name via dhcp
bc777fda45 nic_router: config attribute 'dns_config_from'

m-stein, 2021-08-11:
We could make configuration and report look a bit more elegant regarding the DNS config. Something like:
```
<dns-config domain_name="genodians.org" name_server="1.1.1.1" name_server="8.8.8.8"/>
```
As sub-node for both the ``<dhcp-server>`` configuration tag and the ``<domain>`` report tag.

But I'm unsure whether it's worth breaking the interfaces once more.

m-stein, 2021-08-12:
Added a fixup:
fixup "nic_router: propagate dns domain name via dhcp" (reject oversized names)

chelmuth, 2021-08-12:
> We could make configuration and report look a bit more elegant regarding the DNS config. Something like:
>
> ```
> <dns-config domain_name="genodians.org" name_server="1.1.1.1" name_server="8.8.8.8"/> [...]

I doubt it's worth the cosmetic esp. because this isn't well-formed XML (see https://www.w3.org/TR/xml/#dt-stag "Well-formedness constraint: Unique Att Spec").

m-stein, 2021-08-12:
Ok, so just to get it right in case we later come back to this (as there are more configurable aspects to DNS):
```
<dns-config domain_name="genodians.org">
   <name-server address="1.1.1.1"/>
   <name-server address="8.8.8.8"/>
<dns-config>
```

chelmuth, 2021-08-16:
There seems to be an artificial dependency to `Net::Domain::print()` in *test/nic_router_dhcp/manager* after the last patches This only triggers on riscv.

```
make -C build/riscv/ test/nic_router_dhcp

checking library dependencies...
  Library ldso_so_support
  Library net
  Library ld
  Library base
  Program test/nic_router_dhcp/manager/test-nic_router_dhcp-manager
    COMPILE  main.o
  Program test/nic_router_dhcp/client/test-nic_router_dhcp-client
    LINK     test-nic_router_dhcp-manager
/usr/local/genode/tool/19.05/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: dns.o: in function `void Genode::Output::out_args<char const (&) [2], Net::Domain const&, char const (&) [44], char const*&>(Genode::Output&, char const (&) [2], Net::Domain const&, char const (&) [44], char const*&)':
.../genode/repos/base/include/base/output.h:49: undefined reference to `Net::Domain::print(Genode::Output&) const'
collect2: error: ld returned 1 exit status
.../genode/repos/base/mk/prg.mk:217: recipe for target 'test-nic_router_dhcp-manager' failed
make[2]: *** [test-nic_router_dhcp-manager] Error 1
var/libdeps:113: recipe for target 'test-nic_router_dhcp-manager.prg' failed
make[1]: *** [test-nic_router_dhcp-manager.prg] Error 2
Makefile:301: recipe for target 'gen_deps_and_build_targets' failed
make: *** [gen_deps_and_build_targets] Error 2
```

chelmuth, 2021-08-16:
Great! Merged to staging.

commit e2a79d8a (2021-08-12): nic_router: propagate dns domain name via dhcp

With this commit, the NIC router DHCP client reads out the first DNS domain
name (DHCP option 15) if any from a DHCP reply that generates an IPv4 config
for a domain and stores the name together with the IPv4 config for that domain.
DNS domain names are reported via the new report tag '<dns-domain>' if the
'config' attribute in the config tag '<report>' is set.

Furthermore, the NIC router DHCP server becomes able to obtain a DNS domain
name from another domain that has a DHCP client dynamically (given the config
attribute 'dns_config_from' is se [... cut]
closed 2021-08-18 by chelmuth