genodelabs/genode #4249: Compile error if using `Register::Bitfield::mask()` with 64bit-wide bitfield

jschlatow, 2021-08-16:
...because we are not allowed to do `1 << 64`.

This error occured to me when I declared a Register for keeping `Trace::Timestamp` values. In order to have an architecture-independent binary format of the trace buffer content, I use `uint64_t` as the register's base type but write the timestamp to a Bitfield that is as wide as `Trace::Timestamp`. In case the latter is `uint64_t`, the compiler complains with a shift overflow error.

jschlatow, 2021-08-16:
@m-stein commit d3a0d0c contains the suggested fix and the corresponding test case.

m-stein, 2021-08-17:
@schlatow Thanks for the commit! I'd suggest some small adjustments (see fixup 68f6291e2e). First, I think that the shift width should rather be checked against the register width, so, we prevent errors also for ``access_t`` other than ``uint64_t``. Second, maybe we should consider also bitfield widths greater the register width - although there is no actual use case for that, the framework allows for such a configuration by just cutting the bitfield accordingly. Third, I extended the test a bit to cover also bitfield offsets > 0 and bitfield widths > register width.

What do you think?

jschlatow, 2021-08-17:
Looks good to me. From a readibility perspective, I'd stick to either `_WIDTH` or `WIDTH` though. See [L157+L158](https://github.com/genodelabs/genode/blob/68f6291e2e702cc290afc9910b14bfb37bc16814/repos/base/include/util/register.h#L157).

m-stein, 2021-08-18:
Yeah, you're right, I initially used ``WIDTH`` only but than I had to add a cast because the compared enums were from different type and so I switched to ``_WIDTH`` without adapting the other place. I adapted the fixup accordingly (0a3c4d8f08).

closed 2021-08-31 by chelmuth