genodelabs/genode #4250: tool/run: support new u-boot image format (FIT)

jschlatow, 2021-08-20:
A while ago, u-boot introduced a new image format called FIT (flattened image tree) which simplifies booting a Linux system which typically requires not only a kernel image but also a dtb and a ramdisk. A FIT image combines all ingredients into a single file and adds some metadata like checksums.

Unfortunately, the former uImage format is regarded as legacy. In fact, u-boot can be compiled without support for legacy images.

jschlatow, 2021-08-20:
A very basic variant of a FIT image can be created by adding `-f auto` to the usual mkimage arguments. Commit 3f3af96 adds a image/uboot_fit script that takes care of this and creates an image.itb instead of a uImage.

Note, that booting an image.itb which doesn't contain a dtb may cause u-boot's bootm command to fail. At least this is the case of the pre-built u-boot that I tried on the Zybo board. A workaround for this is to execute the individual boot steps seperately, i.e.

```
bootm start
bootm loados
bootm go
```

This skips the linux-specific preparatory steps that depend on the dtb.

chelmuth, 2021-08-23:
I merged the commit to staging. Do you consider this issue fixed by the commit?

jschlatow, 2021-08-23:
@chelmuth, yes the commit fixes this issue.

closed 2021-08-31 by chelmuth