genodelabs/genode #4256: ccache: avoid use of unsafe direct mode

nfeske, 2021-08-24:
By default, ccache uses the so-called "direct" mode, which sidesteps the C preprocessor for the best performance.

In the presence of shadow headers (as used by our new DDE-Linux), this can cause subtle problems. E.g., in the following sequence of events during the development of a DDE-Linux-based driver:

* Building a driver, populating the ccache
* A new shadow header appears, e.g., lx_emul/shadow/linux/of.h (e.g., when switching development branches)
* A rebuild of the driver will wrongly prompt ccache to hand out cached object files (and .d files) because the new shadow header remains unnoticed by ccache. It would only be detected if running the cpp again.

This can result in sporadic inconsistencies with the developer chasing ghosts for a while until ccache springs in mind a potential source of trouble. To avoid such frustrating experiences, we should force ccache to always operate in the (much safer) preprocessor mode by setting the CCACHE_NODIRECT environment variable.

chelmuth, 2021-08-24:
For reference: I started a discussion at https://github.com/ccache/ccache/discussions/921 if this "unsafe" behavior is expected.

chelmuth, 2021-08-30:
According to ccache maintainer @jrosdahl [direct mode does not support this use case](https://github.com/ccache/ccache/discussions/921#discussioncomment-1248118). The limitation was [already discussed](https://www.mail-archive.com/ccache@lists.samba.org/msg00920.html) back in 2012.

commit a926e5cf (2021-08-24): build: operate ccache in preprocessor mode

This avoids consistency problems when header files are shadowed.

Fixes #4256
commit 82a51d8e (2021-08-30): build: operate ccache in preprocessor mode

This avoids consistency problems when header files are shadowed.

Fixes #4256
closed 2021-08-31 by chelmuth