genodelabs/genode #4258: sftp support in ssh_terminal server component

tomga, 2021-09-13:
Recently I implemented support for SFTP protocol as an add-on to existing `ssh_terminal` component. Implementation rebased on latest release is in my [ssh_sftp](/tomga/genode/commits/ssh_sftp) branch.

Implementation is divided into following commits:
 * fafec7bc0a ssh_terminal: added missing include
 * e67290306d libssh port: required modifications for sftp
 * 24c3695292 libssh port: backported sftp_server_free
 * 289a738f83 libssh port: immediate mode for ssh poll
 * ea389d8e4a sftp: extended ssh_terminal with sftp support
 * df7f1aa455 ssh: add run script to showcase terminal and sftp
 * 73d5f5e210 depot: updated ssh specific depot hashes

Changes to `libssh` port are divided to functional parts and each commit provides short description of introduced change.

Main commit (ea389d8e4a) contains quite extensive description of introduced changes to `ssh_terminal` itself along with design decisions I took when implementing this feature.

Some backward incompatible changes were introduced in `ssh_terminal` configuration that allow to cleanly describe new options. Existing scripts and README file have been updated to reflect those changes.

This work has been sponsored by Gapfruit and with help from people from Gapfruit, especially @sidhussmann and @rite whom I thank for their cooperation.

chelmuth, 2021-09-15:
Thanks for this contribution esp. the clean commit structure. But, before I merge more than ca0ab288a7649367cbe6d17fbae2de2f7f7ff7bf I'd like to discuss some points.

The SSH terminal is a neglected component that was not used at Genode Labs for months beside some thorough IP-protocol/socket debugging sessions by @alex-ab. To be honest, we were thinking about moving the component to world for a long time now. Therefore, I'd like to hear your opinion about this step. libssh on the other will stay in the genode repo for its character as a mandatory protocol library. If you agree with the migration of the terminal to world, I'd appreciate an additional commit that removes the component and test scenarios from genode and adds the current state to world.

>     * [24c3695](https://github.com/genodelabs/genode/commit/24c3695292c9ea784a0e33279d55c099b6cc5311) libssh port: backported sftp_server_free

Could it be an option to update libssh to a recent 0.9 version to prevent patching? I did not look into newer libssh versions and like to hear about your opinion before.

>     * [e672903](https://github.com/genodelabs/genode/commit/e67290306d836d2525670e76cc5feaacf316c990) libssh port: required modifications for sftp
> 
>     * [289a738](https://github.com/genodelabs/genode/commit/289a738f8322b7634918b2b5fce4a7c68881c228) libssh port: immediate mode for ssh poll

Currently, I'm unsure about merging these changes to the library. Aren't there alternative options to use the library (maybe in its version 0.9) unchanged? The background of my question is that I have to rule out the case that bad design decisions for the SSH terminal implementation force us to inadequate changes to libssh.

tomga, 2021-09-15:
> ... If you agree with the migration of the terminal to world, I'd appreciate an additional commit that removes the component and test scenarios from genode and adds the current state to world.

I have no problem with it and I can prepare such patch. Of course this action depends on decision if `libssh` patches are accepted.

> Could it be an option to update libssh to a recent 0.9 version to prevent patching? I did not look into newer libssh versions and like to hear about your opinion before.

Of course update to 0.9.x version would prevent need for this patch. After all it is a backport from later version.

At that time I made a quick attempt to upgrade and it wasn't as straightforward as I hoped for. After finding out that configuration handling in `libssh` was changed (I think it is read from file now) and I would make more changes to existing implementation and existing run scripts I decided to give up and backport this single function. I'd expect that eventual upgrade is not more complicated after adding sftp support than it was before.

> Currently, I'm unsure about merging these changes to the library. Aren't there alternative options to use the library (maybe in its version 0.9) unchanged?

I believe that functionality from those two patches is not available in newer library version. I checked changes in sftp area (and because of that I new about new function for freeing resources) but I did not see any changes that would allow handling sftp from callbacks.

> The background of my question is that I have to rule out the case that bad design decisions for the SSH terminal implementation force us to inadequate changes to libssh.

I believe that design decisions taken for SSH terminal implementation were not bad. I'd call them unfortunate. Decision to have one thread and in consequence handling everything from callbacks is also a decision to use less common and less tested pattern for using `libssh` library. Even existing `event_bind.patch` patch to library was also added to allow non-blocking processing. My patches are needed for exactly the same thing. `sftp_support.patch` allows to handle in non-blocking way existing SFTP functionality (by allowing to replace functions for reading and writing sftp packets) and `event_immediate_poll.patch` allows to prevent blocking flushing and calling callbacks recursively.

Design decision to have one posix thread for each incoming connection would probably simplify interaction with `libssh` library but I'm not sure if it wouldn't bring complications in other areas. I definitely did not want to redesign `ssh_terminal` completely for this implementation.

chelmuth, 2021-09-16:
I agree that the migration to the world repo depends on the decisions regarding libssh.

Also, I'm with you that an update of libssh to 0.9 must be motivated sufficiently and perhaps scheduled for some time in the future. I brought this up only for the reason that API changes like [[1]](https://api.libssh.org/stable/group__libssh__poll.html) could help to prevent more patching of the library. Patches like *event_bind.patch* substantiate my impression that we should not change the library but the component that uses it. @cnuke may contribute more background about the current patch, his past motivation, and current view.

In my opinion the crux of the current design is the intrinsic combination of the Terminal server and the SSH/SFTP server in one component. This fact lead to subsequent decisions regarding threading and local synchronization protocols in the current implementation. IIRC we (as in you and me) had a short discussion about implementing the SFTP server as plain POSIX component, which would make things much less complicated and also enables prototyping on plain Linux. I'd also strongly vote for dropping the Terminal functionality including exec channel if it is no requirement in your scenario. Keep in mind that uploading files via SFTP in combination with the file-watching facility can also trigger operations on the target without the exec channel.

Regarding you suggestion to use multiple threads, I'm not certain about the robustness of multi threading in libssh because of comments like the following [[2]](https://api.libssh.org/stable/libssh_tutor_threads.html).

> You cannot use a single session (or channels for a single session) in several threads at the same time. This will most likely lead to internal state corruption.

tomga, 2021-09-16:
> I brought this up only for the reason that API changes like [[1]](https://api.libssh.org/stable/group__libssh__poll.html) could help to prevent more patching of the library.

Can you explain what API changes you have in mind? I've checked and `include/libssh/poll.h` in `libssh` repository has no changes since 2015.

> In my opinion the crux of the current design is the intrinsic combination of the Terminal server and the SSH/SFTP server in one component. This fact lead to subsequent decisions regarding threading and local synchronization protocols in the current implementation.

Maybe you are right.

On the other hand I know that SFTP handling in `libssh` and especially assembling of SFTP packets is implemented with blocking reads so there is no way to use this code directly without creating separate threads or processes. Direct conclusion from that is that not all library functionality is ready to be used in an event driven way from callbacks. After that you either fix issues with patches as they appear or you redesign component to multithreaded/multiprocess version.

I did not want to start from scratch so I went the first route. We can argue if that was a good decision.

> IIRC we (as in you and me) had a short discussion about implementing the SFTP server as plain POSIX component, which would make things much less complicated and also enables prototyping on plain Linux.

I cannot recall such discussion. Can you point me to it please?

> I'd also strongly vote for dropping the Terminal functionality including exec channel if it is no requirement in your scenario.

It is not a requirement. But I don't see what difference does it make. Unless you think I should implement everything from scratch in different threading/process model.

> Keep in mind that uploading files via SFTP in combination with the file-watching facility can also trigger operations on the target without the exec channel.

I don't see how is this relevant. There is no file watching functionality in SFTP. You can think about it as a remote POSIX compliant file access API. Actually even access flags are almost the same.

> Regarding you suggestion to use multiple threads, I'm not certain about the robustness of multi threading in libssh because of comments like the following [...]

I wrote:

> Design decision to have one posix thread for each incoming connection would probably simplify interaction with `libssh`.

and I meant that in this case you'd have different session for each connection. If I'm wrong about it then definitely this design would not work.

tomga, 2021-09-16:
I looked now at `libssh` open issues and I found some that are relevant to our discussion here. Especially two:

 * [T43: add a ssh_event_add_bind() function](https://gitlab.com/libssh/libssh-mirror/-/issues/78)
 * [T163: Implement fully async sftp read and write](https://gitlab.com/libssh/libssh-mirror/-/issues/65)

chelmuth, 2021-09-20:
Looks like I could not spark your interest in a redesign of the solution to separate the SFTP server from the SSH-terminal server or revise the decision to tightly couple the SSH server with the Genode Terminal server in one component. Therefore I propose the following procedure.

- I review the libssh patches (namely e67290306d836d2525670e76cc5feaacf316c990 24c3695292c9ea784a0e33279d55c099b6cc5311 and 289a738f8322b7634918b2b5fce4a7c68881c228) for a merge to staging.

- You provide me with a ssh_terminal removal commit for genode.git and a commit for inclusion into world.git. Subsequent SFTP adaptation commits (namely ea389d8e4a7408083c81d2f88ffc952869173210 and df7f1aa455e2217236a13517639641f8092fcbfd) can then be merged to world.git directly.

If this meets your approval let me know.

tomga, 2021-09-20:
In short: yes it does.

I think I misunderstood your intentions earlier. It doesn't change my response much though and I'll try to summarize situation.

I think we have two issues which have been mixed in current discussion:

 * requirement for patches to `libssh` port;
 * splitting terminal and sftp functionality.

I think now that that you thought that by addressing the latter we would eliminate the former. I think I showed that it wasn't true. To eliminate the need for patching `libssh` component design would have to be completely changed and I'm afraid besides new implementation it could add new problems to solve.

If we decided to just split sftp functionality we would need to have a common part probably in a form of a library. I looked at it briefly when I implemented this addition but I left this idea as I thought that disabling functionality in configuration was simple enough for now. Possibly if some other ssh subsystem is implemented someday than transition to some "plugin like" architecture could be considered.

tomga, 2021-09-22:
There is one more thing that can be considered. I think that now `ssh_server` would be a better name for this component than `ssh_terminal`. I did not change that name earlier as it could be done later anyway but maybe it should be considered now during transition to genode world? What do you think about it?

If you think it is a good idea than how would you like it do be done? I think that this rename should be already performed before initial commit to genode world so there won't be any renames in history. Would that be ok?

chelmuth, 2021-09-22:
I've no objections - just remove *ssh_terminal* and all references to it (e.g., *ssh_exec_channel.run*) from genode.git and add *ssh_server* to world.git.

commit ecb1a618 (2021-11-29): ssh_terminal: moved to genode world

Fixes #4258
closed 2021-11-30 by chelmuth