genodelabs/genode #4259: Update Pine drivers to Linux 5.14

nfeske, 2021-09-14:
Apparently, the Pine-A64-LTS board can no longer be purchased in its original form because the used Realtek Ethernet PHY chip is no longer available. This prompted the creation of a new board revision (version 2). The new revision uses a MotorComm YT8511 Gigabit Ethernet PHY.

Unfortunately, the new PHY chip is not supported by Linux 5.12, which is used by the current version of the genode-allwinner drivers. Since Linux 5.14 features a driver for this PHY chip, an update of genode-allwinner to Linux 5.14 is needed in order to support the board revision. 

For the record, the new PHY is not supported by U-Boot yet. So network boot is not possible with the new board revision.

nfeske, 2021-09-14:
The issue is fixed by the two commits:

https://github.com/nfeske/genode/commit/af1b1fd060bfc63ba078ccb179d971834fc47278 contains minor adjustments of dde_linux to accommodate kernel-internal interface changes.

https://github.com/nfeske/genode-allwinner/commit/a4ce7ca6036cfbcf172a901a744f795acbec05c1 updates the genode-allwinner repository to Linux 5.14.1

With those two commits, both the EMAC network driver (tested on the original Pine-A64-LTS board) and the DE framebuffer driver (tested on the Pinephone) work with the new kernel version.

@skalk We may consider the following changes in dde_linux down the road:

* The `__alloc_pages_nodemask` function has been replaced by a function called `__alloc_pages_bulk` that has a more sensible interface. Since, `__alloc_pages_nodemask` is implemented in dde_linux/src/lib/lx_emul/shadow/mm/page_alloc.c, we may consider adding the `__alloc_pages_bulk` too. Right now, I'm keeping the implementation local to the genode-allwinner repository.

* The `state` member of `struct task` got renamed to `__state`. I don't know how to deal with this change in the best way. As an interim solution, I'm redefining `-D__state=state` in the genode-allwinner repository.

devoneonone, 2023-10-28:
I have bought one of these cards  PINE A64-LTS V2  & on my board, it is a  chip from  # Realtek  that  do te interface   the port Ethernet,, it 's full name is RTL8211E.  AND AFTER THAT I'M AQUIRING THIS BOARD  THAT WAS DESTINED TO DO GPIO'S INVESTIGATION AND DISCOVERY BUT I DON'T KNOW WHICH  OS CAN HELP ME TO GET ACCESS TO THESE PIN AND I THINK THAT WAS HIS NAME  PINE A64-LTS V2  THAT THAT HAD A ATTRACTION FOR BUY THIS