genodelabs/genode #4260: intel/gpu: Add support for Gen9+

ssumpf, 2021-09-21:
Currently the GPU multiplexer only supports IGT-devices found on Broadwell (Gen8). We want to add support for Skylake (Gen9) and Kabylake (Gen9 and Gen9.5) as well.

ssumpf, 2021-09-21:
The commits above add support for Intel GPU's Gen9/9.5 to Genode.

nfeske, 2021-09-21:
Congratulations @ssumpf, @alex-ab, and @cnuke for solving all these tricky issues!

ssumpf, 2021-09-21:
@nfeske: 3c0e644 uses classes from `commands.h` to setup `MI_*` commands instead of macros.

ssumpf, 2021-09-22:
@nfeske: ccb8037 removes the signal entrypoint from libdrm for Iris. It is not present in entaviv.

ssumpf, 2021-09-24:
@chelmuth: 67f4019 enables building the multiplexer for 32bit platforms.

alex-ab, 2021-09-24:
@chelmuth: fc87fe46faeed15fab06bfbaeb924a8405a4084c implements the discussed feature, to add an additional supported device on Sculpt on the fly

chelmuth, 2021-09-24:
@alex-ab I've no objections to merge the commit beside that I like to ask you to move *gpu_drv.config* into *gpu/intel*. It feels more natural to host this driver specific information in the driver directory like we did with the event-filter chargen files.

alex-ab, 2021-09-27:
@chelmuth: please find the adjusted commit 895647a74a49808e907a6375d70812eacd34a01a which moves the gpu_drv config

nfeske, 2021-10-14:
Fixed in master.