genodelabs/genode #4267: app/sequence does not handle child exit properly

jschlatow, 2021-09-28:
When tracking down a sporadic `Ipc_error` in test-trace (#4261), I found out that the sequence app does not immediately stop the child when it called `parent().exit()` but instead destructs the child asynchronously. This leads to the situation that `parent().exit()` returns and `ld.lib.so` tries to destruct the `Attached_rom_dataspace` for the config rom. In consequence, there is a race condition between sequence and the child: If sequence destructed the rom service first, we get an `Ipc_error`.

chelmuth, 2021-09-28:
What a shame that results from the fact we never adapted the sequence tool to the `Sandbox` API. Many thanks for delving into this.

jschlatow, 2021-09-28:
Commit 390d6c3 eliminates the issue.

nfeske, 2021-11-30:
The commit reached the master branch.

closed 2021-11-30 by nfeske